freechipsproject/chisel-testers #32: AdvTester: Second of two tests fails when run together---each works when run individually

stevenmburns, 2016-09-16:
I've tried to do some parameterized design in chisel3.
I've defined two tests which use testbenches and implemention of the same module but with different parameters. It seems that the implementation in the second test is not elaborated (caused the implementation with different/wrong parameters to be used.)

Here is a complete file that will expose this issue.

````
package bug

import chisel3._
import chisel3.iotesters._
import org.scalatest.GivenWhenThen
import scala.runtime.ScalaRunTime.stringOf

class CarryChainIfc(N:Int,width:Int) extends Module {
   val io = new Bundle {
      val a  = Vec( N, UInt(INPUT,width=width))
      val co = Vec( N, UInt(OUTPUT,width=width))
   }
}

class ORChain(N:Int) extends CarryChainIfc(N,1) {
   require( io.a.length == io.co.length)
   require( io.a.length == N)
   io.co(0) := io.a(0)
   for( ((a,ci),co)<-io.a.toList.tail.zip( io.co).zip( io.co.tail)) {
      co := a | ci
   }
}

abstract class Action
case class ActionInputArray(val nm:String, val s:Array[BigInt])  extends Action
case class ActionOutputArray(val nm:String, val s:Array[BigInt])  extends Action
case class ActionDelay(val v:Int)  extends Action

class CarryChainTests[T <: CarryChainIfc]( ActList:List[Action], c: T) extends AdvTester(c,verbose=true)
{

  for (a <- ActList) {  
    a match {
    case ActionInputArray("a",s) => {
       require( s.length == c.io.a.length)
       for ( (f,a) <- c.io.a.zip( s)) {
          reg_poke( f, a)
       }
    }
    case ActionOutputArray("co",s) => {
       require( s.length == c.io.co.length)
       for ( (f,a) <- c.io.co.zip( s)) {
          expect( f, a)
       }
    }
    case ActionDelay(v) => takesteps(1)()
    }
  }
}

class CarryChainTester[ +T <: CarryChainIfc]( val dut: () => T) extends ChiselFlatSpec with GivenWhenThen {
//  val backend = "firrtl"
  val backend = "verilator"
  val action_buffer= new collection.mutable.ListBuffer[Action]()

  def InputArray( nm:String, tag:String, s:Array[BigInt]):String = {
     action_buffer += new ActionInputArray(nm,s)
     s"InputArray $nm $tag ${stringOf(s)}"
  }

  def OutputArray( nm:String, tag:String, s:Array[BigInt]):String = {
     action_buffer += new ActionOutputArray(nm,s)
     s"OutputArray $nm $tag ${stringOf(s)}"
  }
[... 205 lines cut ...]
[info]   + And After 1 cycle 
[info]   + Then OutputArray co is Array(1, 1, 1, 1) 
[info]   + When InputArray a is Array(0, 1, 0, 0) 
[info]   + And After 1 cycle 
[info]   + Then OutputArray co is Array(0, 1, 1, 1) 
[info]   + When InputArray a is Array(0, 0, 1, 0) 
[info]   + And After 1 cycle 
[info]   + Then OutputArray co is Array(0, 0, 1, 1) 
[info]   + When InputArray a is Array(0, 0, 0, 1) 
[info]   + And After 1 cycle 
[info]   + Then OutputArray co is Array(0, 0, 0, 1) 
[info] ScalaCheck
[info] Passed: Total 0, Failed 0, Errors 0, Passed 0
[info] ScalaTest
[info] Run completed in 11 seconds, 89 milliseconds.
[info] Total number of tests run: 2
[info] Suites: completed 2, aborted 0
[info] Tests: succeeded 1, failed 1, canceled 0, ignored 0, pending 0
[info] *** 1 TEST FAILED ***
[error] Failed: Total 2, Failed 1, Errors 0, Passed 1
[error] Failed tests:
[error] 	bug.ORChainTest4
[error] (test:test) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 14 s, completed Sep 16, 2016 3:03:55 PM
````

stevenmburns, 2016-09-20:
Should we just not use AdvTester and PeekPokeTester anymore? We discovered another issue and I wonder if we should bother reporting it.

stevenmburns, 2016-09-21:
I added the other issue just now.

stevenmburns, 2016-09-22:
@ucbjrl @chick This one is also interesting to us. I think the same issue is there for a standard PeekPokeTester, not just for AdvTester.

donggyukim, 2016-09-22:
I assume this happens when you run testers with the same name modules because Driver.run generates the target directories according to the module names. Why don't you try chiselMainTest with "--targetDir"? I hope this should work.

stevenmburns, 2016-09-23:
I'm not using chiselMainTest. Is there a way to do that within the
scalatest "ChiselFlatSpec"?

This works in SteppedHWIOTester because it creates these directories in
/tmp. Is it possible to make it work the same way for PeekPokeTester.

PR #40 (2016-10-07): create temp directory to prevent thread unsaftety
I hope this resolve #32. Driver creates a temp dir unless we give one.
closed 2016-10-07 by ucbjrl