freechipsproject/chisel-testers #34: PeekPokeTester: including simulation time assert (or stop) causes verilator compilation error

stevenmburns, 2016-09-21:
We're trying to place runtime asserts in mocked implementation code. The following idea works using firrtl-interpreter and SteppedHWIOTester, but not with the PeekPokeTester due to a missing symbol.

````
g++    myRisc-harness.o verilated.o verilated_vcd_c.o VmyRisc__ALL.a    -o VmyRisc -lm -lstdc++  2>&1 | c++filt
VmyRisc__ALL.a(VmyRisc__ALLcls.o): In function `VmyRisc::_sequent__TOP__3(VmyRisc__Syms*)':
VmyRisc__ALLcls.cpp:(.text+0x1ca): undefined reference to `sc_time_stamp()'
collect2: error: ld returned 1 exit status
````

Here is the source code and the complete log file.

````
package myRisc

import chisel3._

class mockRegFileOut() extends Module{
  val io = new Bundle {
    val addr = Bits(INPUT, 8)
    val DataO = Bits(OUTPUT, 32)
  }
  
  assert(io.addr === UInt(0),"ASSERT:Read at the wrong Register")
  
  when(io.addr === UInt(0)){
    io.DataO := UInt(128)
  }
}

class myRisc extends Module{
  val io = new Bundle {
    val wrData = Bits(INPUT, 32)

    val valid  = Bool(OUTPUT)
    val out    = Bits(OUTPUT, 32)
  }
  
  val inst = io.wrData
  
  
  //instantiate mock
  val mock = Module(new mockRegFileOut).io
  mock.addr := UInt(16)//inst(15, 8)
// Comment the next line to see a mocking failure
  mock.addr := inst(15, 8)
  
  
  val rci  = inst(23,16) 
  
  when (rci === UInt(255)){
    io.valid := Bool(true)
    io.out := mock.DataO
  }.otherwise{
    io.valid := Bool(false)
  }
  
}

import org.scalatest.{ Matchers, FlatSpec, GivenWhenThen}

import chisel3.iotesters._

class myRiscPeekPokeTester( val c:myRisc) extends PeekPokeTester(c) {
   poke( c.io.wrData, (0<<24)+(255<<16))
   step(1)
   expect( c.io.valid, 1)
   expect( c.io.out, 128)
}

class myRiscPeekPokeTest extends FlatSpec with Matchers {
  behavior of "myRiscPeekPokeTest"

  it should s"work" in {
    chisel3.iotesters.Driver((() => new myRisc),"verilator") { c => new myRiscPeekPokeTester(c) } should be (true)
  }  
}
````

The output of "sbt test":
````
> test
[info] [0.002] Elaborating design...
[info] [0.073] Done elaborating.
verilator --cc myRisc.v --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --trace -O2 --top-module myRisc +define+TOP_TYPE=VmyRisc +define+PRINTF_COND=!myRisc.reset +define+STOP_COND=!myRisc.reset -CFLAGS -Wno-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h -Mdir test_run_dir/myRisc.myRisc --exe myRisc-harness.cpp
make: Entering directory `/nfs/site/disks/scl.work.59/ppt/smburns/tdd_using_chisel-testing_flows/bugs/bug-sc_time_stamp/test_run_dir/myRisc.myRisc'
g++  -I.  -MMD -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h   -c -o myRisc-harness.o myRisc-harness.cpp
g++  -I.  -MMD -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h   -c -o verilated.o /nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/verilated.cpp
g++  -I.  -MMD -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h   -c -o verilated_vcd_c.o /nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/verilated_vcd_c.cpp
/usr/intel/bin/perl /nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VmyRisc.cpp > VmyRisc__ALLcls.cpp
/usr/intel/bin/perl /nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VmyRisc__Trace.cpp VmyRisc__Syms.cpp VmyRisc__Trace__Slow.cpp > VmyRisc__ALLsup.cpp
g++  -I.  -MMD -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h   -c -o VmyRisc__ALLsup.o VmyRisc__ALLsup.cpp
g++  -I.  -MMD -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h   -c -o VmyRisc__ALLcls.o VmyRisc__ALLcls.cpp
      Archiving VmyRisc__ALL.a ...
ar r VmyRisc__ALL.a VmyRisc__ALLcls.o VmyRisc__ALLsup.o
ranlib VmyRisc__ALL.a
g++    myRisc-harness.o verilated.o verilated_vcd_c.o VmyRisc__ALL.a    -o VmyRisc -lm -lstdc++  2>&1 | c++filt
VmyRisc__ALL.a(VmyRisc__ALLcls.o): In function `VmyRisc::_sequent__TOP__3(VmyRisc__Syms*)':
VmyRisc__ALLcls.cpp:(.text+0x1ca): undefined reference to `sc_time_stamp()'
collect2: error: ld returned 1 exit status
make: Leaving directory `/nfs/site/disks/scl.work.59/ppt/smburns/tdd_using_chisel-testing_flows/bugs/bug-sc_time_stamp/test_run_dir/myRisc.myRisc'
[info] myRiscPeekPokeTest:
[info] myRiscPeekPokeTest
[info] - should work *** FAILED ***
[info]   java.lang.IllegalArgumentException: requirement failed: test_run_dir/myRisc.myRisc/VmyRisc doesn't exists
[info]   at scala.Predef$.require(Predef.scala:219)
[info]   at chisel3.iotesters.TesterProcess$.apply(PeekPokeTesterUtils.scala:112)
[info]   at chisel3.iotesters.SimApiInterface.<init>(SimApiInterface.scala:278)
[info]   at chisel3.iotesters.VerilatorBackend.<init>(VerilatorBackend.scala:290)
[info]   at chisel3.iotesters.setupVerilatorBackend$.apply(VerilatorBackend.scala:282)
[info]   at chisel3.iotesters.Driver$.apply(Driver.scala:22)
[info]   at myRisc.myRiscPeekPokeTest$$anonfun$4.apply$mcV$sp(t.scala:62)
[info]   at myRisc.myRiscPeekPokeTest$$anonfun$4.apply(t.scala:62)
[info]   at myRisc.myRiscPeekPokeTest$$anonfun$4.apply(t.scala:62)
[info]   at org.scalatest.Transformer$$anonfun$apply$1.apply$mcV$sp(Transformer.scala:22)
[info]   ...
[info] ScalaCheck
[info] Passed: Total 0, Failed 0, Errors 0, Passed 0
[info] ScalaTest
[info] Run completed in 3 seconds, 1 milliseconds.
[info] Total number of tests run: 1
[info] Suites: completed 1, aborted 0
[info] Tests: succeeded 0, failed 1, canceled 0, ignored 0, pending 0
[info] *** 1 TEST FAILED ***
[error] Failed: Total 1, Failed 1, Errors 0, Passed 0
[error] Failed tests:
[error]         myRisc.myRiscPeekPokeTest
[error] (test:test) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 3 s, completed Sep 21, 2016 11:50:06 AM
````

ucbjrl, 2016-09-22:
The verilator backend should output the following in the dut-harness.cpp file if a simulation time assert has been generated:

    double sc_time_stamp () { return main_time; }

stevenmburns, 2016-09-22:
It seems to not be doing that for me.
````
dlxc1340> egrep sc_t test_run_dir/myRisc.myRisc/myRisc-harness.cpp
dlxc1340> 
````

Here is the complete file:

````
cat test_run_dir/myRisc.myRisc/myRisc-harness.cpp
#include "VmyRisc.h"
#include "verilated.h"
#include "veri_api.h"
#if VM_TRACE
#include "verilated_vcd_c.h"
#endif
#include <iostream>
class myRisc_api_t: public sim_api_t<VerilatorDataWrapper*> {
public:
    myRisc_api_t(VmyRisc* _dut) {
        dut = _dut;
        main_time = 0L;
        is_exit = false;
#if VM_TRACE
        tfp = NULL;
#endif
    }
    void init_sim_data() {
        sim_data.inputs.clear();
        sim_data.outputs.clear();
        sim_data.signals.clear();
        sim_data.inputs.push_back(new VerilatorIData(&(dut->io_wrData)));
        sim_data.outputs.push_back(new VerilatorIData(&(dut->io_out)));
        sim_data.outputs.push_back(new VerilatorCData(&(dut->io_valid)));
        sim_data.signals.push_back(new VerilatorCData(&(dut->reset)));
        sim_data.signal_map["myRisc.reset"] = 0;
    }
#if VM_TRACE
     void init_dump(VerilatedVcdC* _tfp) { tfp = _tfp; }
#endif
    inline bool exit() { return is_exit; }
private:
    VmyRisc* dut;
    bool is_exit;
    vluint64_t main_time;
#if VM_TRACE
    VerilatedVcdC* tfp;
#endif
    virtual inline size_t put_value(VerilatorDataWrapper* &sig, uint64_t* data, bool force=false) {
        return sig->put_value(data);
    }
    virtual inline size_t get_value(VerilatorDataWrapper* &sig, uint64_t* data) {
        return sig->get_value(data);
    }
    virtual inline size_t get_chunk(VerilatorDataWrapper* &sig) {
        return sig->get_num_words();
    } 
    virtual inline void reset() {
        dut->reset = 1;
        step();
    }
    virtual inline void start() {
        dut->reset = 0;
    }
    virtual inline void finish() {
        dut->eval();
        is_exit = true;
    }
    virtual inline void step() {
        dut->clk = 0;
        dut->eval();
#if VM_TRACE
        if (tfp) tfp->dump(main_time);
#endif
        main_time++;
        dut->clk = 1;
        dut->eval();
#if VM_TRACE
        if (tfp) tfp->dump(main_time);
#endif
        main_time++;
    }
    virtual inline void update() {
        dut->_eval_settle(dut->__VlSymsp);
    }
};
int main(int argc, char **argv, char **env) {
    Verilated::commandArgs(argc, argv);
    VmyRisc* top = new VmyRisc;
    std::string vcdfile = "test_run_dir/myRisc.myRisc/myRisc.vcd";
    std::vector<std::string> args(argv+1, argv+argc);
    std::vector<std::string>::const_iterator it;
    for (it = args.begin() ; it != args.end() ; it++) {
      if (it->find("+waveform=") == 0) vcdfile = it->c_str()+10;
    }
#if VM_TRACE
    Verilated::traceEverOn(true);
    VL_PRINTF("Enabling waves..");
    VerilatedVcdC* tfp = new VerilatedVcdC;
    top->trace(tfp, 99);
    tfp->open(vcdfile.c_str());
#endif
    myRisc_api_t api(top);
    api.init_sim_data();
    api.init_channels();
#if VM_TRACE
    api.init_dump(tfp);
#endif
    while(!api.exit()) api.tick();
#if VM_TRACE
    if (tfp) tfp->close();
    delete tfp;
#endif
    delete top;
    exit(0);
}
````

stevenmburns, 2016-09-22:
I tried the patch and it doesn't work on my example (a new error).
These seem to be the relevant errors:
````
myRisc-harness.cpp: In member function ‘virtual void myRisc_api_t::step()’:
myRisc-harness.cpp:63:14: error: ‘class VmyRisc’ has no member named ‘clk’
myRisc-harness.cpp:69:14: error: ‘class VmyRisc’ has no member named ‘clk’
````

Here is the complete log:
````
dlxc1340> sbt test
[info] Set current project to SCL PPT Testing Experiments (in build file:/nfs/site/disks/scl.work.59/ppt/smb
urns/tdd_using_chisel-testing_flows/bugs/bug-sc_time_stamp/)
[info] [0.002] Elaborating design...
[info] [0.093] Done elaborating.
verilator --cc myRisc.v --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --trace -O2 --top-module myRisc +define+
TOP_TYPE=VmyRisc +define+PRINTF_COND=!myRisc.reset +define+STOP_COND=!myRisc.reset -CFLAGS -Wno-undefined-bo
ol-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h -Mdir test_run_dir/myRisc.myRisc --exe myRisc-harnes
s.cpp
make: Entering directory `/nfs/site/disks/scl.work.59/ppt/smburns/tdd_using_chisel-testing_flows/bugs/bug-sc
_time_stamp/test_run_dir/myRisc.myRisc'
g++  -I.  -MMD -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/in
clude -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/vlt
std -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign
-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno
-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h   -c -o myRisc-harness.o myRisc-harness
.cpp
g++  -I.  -MMD -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/in
clude -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/vlt
std -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign
-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno
-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h   -c -o verilated.o /nfs/site/disks/scl
.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/verilated.cpp
g++  -I.  -MMD -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/in
clude -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/vlt
std -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign
-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno
-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h   -c -o verilated_vcd_c.o /nfs/site/dis
ks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/verilated_vcd_c.cpp
/usr/intel/bin/perl /nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator
/bin/verilator_includer -DVL_INCLUDE_OPT=include VmyRisc.cpp > VmyRisc__ALLcls.cpp
/usr/intel/bin/perl /nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator
/bin/verilator_includer -DVL_INCLUDE_OPT=include VmyRisc__Trace.cpp VmyRisc__Syms.cpp VmyRisc__Trace__Slow.c
pp > VmyRisc__ALLsup.cpp
g++  -I.  -MMD -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/in
clude -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/vlt
std -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign
-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno
-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h   -c -o VmyRisc__ALLsup.o VmyRisc__ALLs
up.cpp
g++  -I.  -MMD -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/in
clude -I/nfs/site/disks/scl.work.59/ppt/smburns/chisel/chisel3/verilator_install/share/verilator/include/vlt
std -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign
-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno
-undefined-bool-conversion -O2 -DTOP_TYPE=VmyRisc -include VmyRisc.h   -c -o VmyRisc__ALLcls.o VmyRisc__ALLc
ls.cpp
myRisc-harness.cpp: In member function ‘virtual void myRisc_api_t::step()’:
myRisc-harness.cpp:63:14: error: ‘class VmyRisc’ has no member named ‘clk’
myRisc-harness.cpp:69:14: error: ‘class VmyRisc’ has no member named ‘clk’
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-undefined-bool-conversion" [enabled by default]
cc1plus: warning: unrecognized command line option "-Wno-parentheses-equality" [enabled by default]
make: *** [myRisc-harness.o] Error 1
make: *** Waiting for unfinished jobs....
make: Leaving directory `/nfs/site/disks/scl.work.59/ppt/smburns/tdd_using_chisel-testing_flows/bugs/bug-sc_
time_stamp/test_run_dir/myRisc.myRisc'
[info] myRiscPeekPokeTest:
[info] myRiscPeekPokeTest
[info] - should work *** FAILED ***
[info]   java.lang.IllegalArgumentException: requirement failed: test_run_dir/myRisc.myRisc/VmyRisc doesn't 
[... 2 lines cut ...]
[info]   at chisel3.iotesters.TesterProcess$.apply(PeekPokeTesterUtils.scala:112)
[... cut ...]
[info] Run completed in 3 seconds, 202 milliseconds.
[info] Total number of tests run: 1
[info] Suites: completed 1, aborted 0
[info] Tests: succeeded 0, failed 1, canceled 0, ignored 0, pending 0
[info] *** 1 TEST FAILED ***
[error] Failed: Total 1, Failed 1, Errors 0, Passed 0
[error] Failed tests:
[error]         myRisc.myRiscPeekPokeTest
[error] (test:test) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 5 s, completed Sep 22, 2016 11:00:19 AM
````
Prehaps I need to mix in a different fix branch.

ucbjrl, 2016-09-22:
Chisel3 is in the process of renaming the clock signals and the testers haven't caught up, Try with commit:

	dda64c1dee16b5da15ac690bd3cd6759c3d5c032

of chisel3, or wait for the chisel-testers branch to catch up.

ucbjrl, 2016-09-22:
Closed by #35.

closed 2016-09-22 by ucbjrl