freechipsproject/chisel-testers #60: Sequential Verilator tests fail and spews data in the root folder

edwardcwang, 2016-11-09:
#32 is supposedly fixed, but in the latest revision (50224d0afb45bcfbf5c4a9a4dd2fd2924ec88ddf) this error seems to have come up again, and instead of using a test_run_dir it just puts all the generated files in the root of the repo:
```
# ls
build.sbt  build.sh  c.py  log.txt  project  README.md  scalastyle-config.xml  scalastyle-test-config.xml  src  target 
# sbt "test-only utils.test.DebouncerTester"

[...first test runs successfully...]

verilator --cc Debouncer.v --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --trace -O0 --top-module Debouncer +define+TOP_TYPE=VDebouncer +define+PRINTF_COND=!Debouncer.reset +define+STOP_COND=!Debouncer.reset -CFLAGS -Wno-undefined-bool-conversion -O0 -DTOP_TYPE=VDebouncer -include VDebouncer.h -Mdir . --exe Debouncer-harness.cpp
make: Entering directory `/home/cc/chisel-template'
g++  -I.  -MMD -I/home/cc/verilator/include -I/home/cc/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wno-undefined-bool-conversion -O0 -DTOP_TYPE=VDebouncer -include VDebouncer.h   -c -o Debouncer-harness.o Debouncer-harness.cpp
Debouncer-harness.cpp: In member function ‘void Debouncer_api_t::init_sim_data()’:
Debouncer-harness.cpp:23:61: error: ‘class VDebouncer’ has no member named ‘io_in_1’
         sim_data.inputs.push_back(new VerilatorCData(&(dut->io_in_1)));
                                                             ^
Debouncer-harness.cpp:24:61: error: ‘class VDebouncer’ has no member named ‘io_in_2’
         sim_data.inputs.push_back(new VerilatorCData(&(dut->io_in_2)));
                                                             ^
Debouncer-harness.cpp:25:61: error: ‘class VDebouncer’ has no member named ‘io_in_3’
         sim_data.inputs.push_back(new VerilatorCData(&(dut->io_in_3)));
                                                             ^
Debouncer-harness.cpp:26:61: error: ‘class VDebouncer’ has no member named ‘io_in_4’
         sim_data.inputs.push_back(new VerilatorCData(&(dut->io_in_4)));
                                                             ^
# ls
build.sbt              Debouncer.v            scalastyle-test-config.xml  VDebouncer__ALLcls.cpp  VDebouncer_classes.mk     VDebouncer__Syms.cpp         veri_api.h
build.sh               Debouncer.vcd          sim_api.h                   VDebouncer__ALLcls.d    VDebouncer.cpp            VDebouncer__Syms.h           verilated.d
c.py  log.txt                src                         VDebouncer__ALLcls.o    VDebouncer_Debouncer.cpp  VDebouncer__Trace.cpp        verilated.o
Debouncer-harness.cpp  project                target                      VDebouncer__ALLsup.cpp  VDebouncer_Debouncer.h    VDebouncer__Trace__Slow.cpp  verilated_vcd_c.d
Debouncer-harness.d    README.md              VDebouncer                  VDebouncer__ALLsup.d    VDebouncer.h              VDebouncer__ver.d            verilated_vcd_c.o
Debouncer-harness.o    scalastyle-config.xml  VDebouncer__ALL.a           VDebouncer__ALLsup.o    VDebouncer.mk             VDebouncer__verFiles.dat

```

Here is the relevant portion of the test code:

```
class DebouncerTester extends ChiselFlatSpec {
  behavior of "Debouncer"

  def get_options(backend:String): TesterOptionsManager = {
    val options = new TesterOptionsManager{
      interpreterOptions = interpreterOptions.copy(setVerbose = false)
      testerOptions = testerOptions.copy(backendName = backend)
    }
    options
  }

  val backendNames = Array[String]("verilator")
  for ( backendName <- backendNames ) {
    it should s"do a basic 6 count (with ${backendName})" in {
      Driver.execute(() => new Debouncer(2, samples=6, percent_stable=5f/6), get_options(backendName)) { c =>
        new DebouncerBasic6Count(c)
      } should be(true)
    }

    it should s"stay high with n=1 (with ${backendName})" in {
      Driver.execute(() => new Debouncer(1, samples=10), get_options(backendName)) { c =>
        new DebouncerStayHigh(c)
      } should be(true)
    }

    it should s"stay high with n=100 (with ${backendName})" in {
      Driver.execute(() => new Debouncer(100, samples=10), get_options(backendName)) { c =>
        new DebouncerStayHigh(c)
      } should be(true)
    }
  }
}
```

edwardcwang, 2016-11-09:
I do note that the fix got moved/rewritten/overwritten in ed1a1d40774811e6934f67ef1655b4bce65fecc9 - could that be the cause of it?

edwardcwang, 2016-12-05:
Seems to work as of c760dd94c5553a5b43a84275225cb5338b3057fd

closed 2016-12-05 by edwardcwang

chick, 2017-05-08:
I've add this to https://github.com/ucb-bar/chisel3/issues/551 Tester Unification.