freechipsproject/chisel-testers #73: Get rid of the test boilerplate

ducky64, 2016-12-14:
This is what we have right now
```
class MyDUT extends Module {
  ... instantiate MyDevice here ...
}

class MyDUTTester(val c: MyDUT) extends PeekPokeTester(c) {
  expect(c.io.blah, someNumber)
  ...
}

class MyDeviceSpec extends ChiselFlatSpec {
  "MyDevice" should "blah" in {
    Driver(() => new MyDUT) {
      c => new MyDUTTester(c)
    } should be (true)
  }
}
```

The amount of repetition there is pretty insane, and even worse is that the Tester logic is its own class, but is dependent on the parameterization that is specified externally in the Spec. Individual expect failures also aren't registered by the test framework (which can automatically generate nice error messages at the invocation site, I think including a stack trace); instead the test framework only tells your test failed somewhere. (yes, the default verbose failures messages helps, but that's a workaround at best, and for whatever reason isn't showing up on Verilator tests)

I think ideally we would have something like:
```
class MyDeviceSpec extends ChiselPeekPokeSpec {
  "MyDevice" should "blah" in {
    val t, c = createTester(new MyDUT)  // essentially creates a test fixture, return an emulator link and the elaborated module for IO access
    t.expect(c.io.blah, someNumber)
   ...
  }
}
```

Much more compact, less repetition, less boilerplate, and most importantly makes writing tests fun for everyone!

Thoughts?

edwardcwang, 2016-12-15:
What goes into `createTester()`?

ducky64, 2016-12-15:
That's the part I'm going to figure out. I think it would invoke the Driver (elaborate and either run Verilator or the firrtl interpreter) and provide an interface layer to the Driver.

chick, 2016-12-15:
I'd like to see expect work more like a standard scala matcher type test, though that may be easy enough with a peek
I think we should also make it easier to access common debugging features.   Things like backends, logging/verbose modes.