freechipsproject/chisel-testers #84: Different behavior from Chisel2 for negative peeks

shunshou, 2017-01-17:
There's no signed conversion, so everything comes out as unsigned BigInts, I believe [tho I'm testing w/ something that extends PeekPokeTester, but I believe this is the current behavior]. For Dsp checks, this kind of matters. I can either do the signed conversion here [to be consistent with Chisel2] or within my own DspTester, but doing it here is somewhat of an API change for people. Does anyone have any opinions? If I do it at a higher level, I feel like I'm just adding patch upon patch to things... but I also need some of these changes ASAP.

@ducky64 @chick @donggyukim /whoever else might be in charge of testers...

chick, 2017-01-17:
@shunshou, we are talking about peeking signals here, right?  It looks to me that the interpreter handles this correctly but that the verilator back end does not. If this is write then I'll assert that we should fix this, in chisel-testers, so verilator backend returns signed values when the signal being peeked or expected is an SInt

shunshou, 2017-01-17:
Yes, peeking. I think poking always is signed when it gets down to the backend but peeking returns unsigned BigInt that needs casting depending on the datatype. Ok, I've been strictly using Verilator. 

But it's also not just an SInt, right? You'd need to do the same for FixedPoint, possibly DspReal(?) [and any future signed types...] since it's not a subclass of SInt and I assume the Verilator backend is just matching off of T <: Data types.

chick, 2017-01-17:
I have just made a PR #85, probably needs comments from chiselers