freechipsproject/chisel-testers #87: Verilator backend breaks for multiple tests

shunshou, 2017-01-23:
```
behavior of "simple module io + lits"

  it should "read lits with gen = sint properly" in {
    dsptools.Driver.execute(() => new SimpleModule(p.genShortS, p.genLongS, includeR = true, p), optionsPass) { c =>
      new LitTester(c)
    } should be (true)
  }

  it should "read lits with gen = fixed properly" in {
    dsptools.Driver.execute(() => new SimpleModule(p.genShortF, p.genLongF, includeR = true, p), optionsPass) { c =>
      new LitTester(c)
    } should be (true)
  }
```

When FIRRTL interpreter is used, both tests pass. When Verilator is used, the second test starts to run, but then fails with

```
Enabling waves..
chisel3.iotesters.TestApplicationException: test application exit - exit code 139
	at chisel3.iotesters.SimApiInterface.throwExceptionIfDead(SimApiInterface.scala:91)
	at chisel3.iotesters.SimApiInterface.chisel3$iotesters$SimApiInterface$$mwhile(SimApiInterface.scala:100)
	at chisel3.iotesters.SimApiInterface.update(SimApiInterface.scala:194)
	at chisel3.iotesters.SimApiInterface.peek(SimApiInterface.scala:281)
	at chisel3.iotesters.VerilatorBackend.peek(VerilatorBackend.scala:336)
	at dsptools.VerboseDspTester.peek(VerboseDspTester.scala:100)
```

If I comment out and run individual tests by themselves, things are fine... So there's something wrong with running Verilator twice? i.e. it shouldn't throw an exception when the test application "dies," because the first one actually did finish? 

Full test code @ https://github.com/ucb-art/Chisel3DSPExample/blob/master/src/test/scala/UsefulExamples/SimpleTBwGenTypeOption.scala

It's hard for me to figure out what should actually be done in this case, since throwExceptionIfDead(exitValue) was put there for some reason???

@chick any ideas?

chick, 2017-01-23:
@shunshou this is a guess.  Scalatest runs tests like this in parallel.  If these two tests are being built in the same directory, then they could be stomping on each other.  There's away to make tests run serially, I think you have to add
parallelExecution in Test := false
in sbt.  @ucbjrl knows about this I think

shunshou, 2017-01-23:
My question is then -- why is it that the two tests in ParameterizedAdderSpec get made into separate directories but this one doesn't? How can I force all tests to be run in separate directories? [or rather, do you even want that to happen?]

shunshou, 2017-01-24:
@chick did this magically get fixed? Or am I just getting lucky with parallel stuff? (It's not crashing atm, even though only 1 directory is generated...).

chick, 2017-01-25:
I think it is luck, I didn't do anything to fix this.