freechipsproject/chisel-testers #94: Post-synthesis simulation

jcmartin, 2017-01-30:
The basic issue is that I need to do synthesis post-simulation. The old way of doing this was to generate the harness from chisel, use VCS to compile it with the post-synthesis verilog and standard cells library, and use the resulting executable (usually simv) back with the chisel infrastructure.

There are two issues one major and one minor.

Minor: There is no way to generate the harness in a known path directory, and you have to use one of the backends to get it to work even when you do not want to run vcs or verilator.

Major: Test-command is unusable.

chick, 2017-01-30:
@jcmartin , will get Major issue ASAP, but with regards to the minor issue.  You can generate the harness in a known directory.  Try, for example:
```scala
    val args = Array("--backend-name", "firrtl", "--fint-write-vcd", "--target-dir", "foo", "--top-name", "RealGCD2")
    iotesters.Driver.execute(args, () => new RealGCD2) { c =>
      new GCDPeekPokeTester(c)
    } should be (true)
```
or more directly, bit not necessarily clearer
```scala
    val manager = new TesterOptionsManager {
      testerOptions = testerOptions.copy(backendName = "firrtl")
      commonOptions = commonOptions.copy(targetDirName = "foo",  topName = "RealGCD2")
      interpreterOptions = interpreterOptions.copy(setVerbose = false, writeVCD = true)
    }
    iotesters.Driver.execute(() => new RealGCD2, manager) { c =>
      new GCDPeekPokeTester(c)
    } should be (true)

```
These will both place the generated stuff in foo.  It is a little quirky in the following way.
**If you do not set the top name, then you will end up with a subdirectory in foo which is the dut
name plus a hashCode**

jcmartin, 2017-01-30:
@chick 

Your example doesn't work. It still runs the FIRRTL interpreter, and it doesn't generate the verilog harness necessary for VCS.

ucbjrl, 2017-01-30:
@chick - I believe they're using the "released" 3.0-SNAPSHOT of chisel3 and the "released" 1.1-SNAPSHOT of chisel.iotesters. We'll need to push updated versions of this of publish local updates.

chick, 2017-01-30:
@ucbjrl I'm at BWRC all afternoon, can we talk about this in morning.

ucbjrl, 2017-01-30:
Absolutely.

chick, 2017-01-31:
@jcmartin I have made a pull request PR #95 
can you test.  Example use is in DriverSpec.scala

jcmartin, 2017-02-02:
I'm trying to use the chisel-testers from master branch in my project, but I'm running into issues of trying to find snapshots that don't exist.

[info] 	[SUCCESSFUL ] org.scala-lang#scala-reflect;2.11.2!scala-reflect.jar (794ms)
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#chisel3_2.11;3.1-SNAPSHOT: not found
[warn] 	:: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found
[warn] 	:: edu.berkeley.cs#firrtl-interpreter_2.11;1.1-SNAPSHOT: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		edu.berkeley.cs:chisel3_2.11:3.1-SNAPSHOT (/home/aa/ugrad/jmartin/.sbt/0.13/staging/facb567f3da2c82e7547/chisel-testers/build.sbt#L15-18)
[warn] 		  +- edu.berkeley.cs:chisel-iotesters_2.11:1.2-SNAPSHOT
[warn] 		edu.berkeley.cs:firrtl_2.11:1.1-SNAPSHOT (/home/aa/ugrad/jmartin/.sbt/0.13/staging/facb567f3da2c82e7547/chisel-testers/build.sbt#L15-18)
[warn] 		  +- edu.berkeley.cs:chisel-iotesters_2.11:1.2-SNAPSHOT
[warn] 		edu.berkeley.cs:firrtl-interpreter_2.11:1.1-SNAPSHOT (/home/aa/ugrad/jmartin/.sbt/0.13/staging/facb567f3da2c82e7547/chisel-testers/build.sbt#L15-18)
[warn] 		  +- edu.berkeley.cs:chisel-iotesters_2.11:1.2-SNAPSHOT
sbt.ResolveException: unresolved dependency: edu.berkeley.cs#chisel3_2.11;3.1-SNAPSHOT: not found
unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found
unresolved dependency: edu.berkeley.cs#firrtl-interpreter_2.11;1.1-SNAPSHOT: not found
	at sbt.IvyActions$.sbt$IvyActions$$resolve(IvyActions.scala:291)
	at sbt.IvyActions$$anonfun$updateEither$1.apply(IvyActions.scala:188)
	at sbt.IvyActions$$anonfun$updateEither$1.apply(IvyActions.scala:165)
	at sbt.IvySbt$Module$$anonfun$withModule$1.apply(Ivy.scala:155)
	at sbt.IvySbt$Module$$anonfun$withModule$1.apply(Ivy.scala:155)
	at sbt.IvySbt$$anonfun$withIvy$1.apply(Ivy.scala:132)
	at sbt.IvySbt.sbt$IvySbt$$action$1(Ivy.scala:57)
	at sbt.IvySbt$$anon$4.call(Ivy.scala:65)
	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:93)
	at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:78)
	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:97)
	at xsbt.boot.Using$.withResource(Using.scala:10)
	at xsbt.boot.Using$.apply(Using.scala:9)
	at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:58)
	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:48)
	at xsbt.boot.Locks$.apply0(Locks.scala:31)
	at xsbt.boot.Locks$.apply(Locks.scala:28)
	at sbt.IvySbt.withDefaultLogger(Ivy.scala:65)
	at sbt.IvySbt.withIvy(Ivy.scala:127)
	at sbt.IvySbt.withIvy(Ivy.scala:124)
	at sbt.IvySbt$Module.withModule(Ivy.scala:155)
	at sbt.IvyActions$.updateEither(IvyActions.scala:165)
	at sbt.Classpaths$$anonfun$sbt$Classpaths$$work$1$1.apply(Defaults.scala:1369)
	at sbt.Classpaths$$anonfun$sbt$Classpaths$$work$1$1.apply(Defaults.scala:1365)
	at sbt.Classpaths$$anonfun$doWork$1$1$$anonfun$87.apply(Defaults.scala:1399)
	at sbt.Classpaths$$anonfun$doWork$1$1$$anonfun$87.apply(Defaults.scala:1397)
	at sbt.Tracked$$anonfun$lastOutput$1.apply(Tracked.scala:37)
	at sbt.Classpaths$$anonfun$doWork$1$1.apply(Defaults.scala:1402)
	at sbt.Classpaths$$anonfun$doWork$1$1.apply(Defaults.scala:1396)
	at sbt.Tracked$$anonfun$inputChanged$1.apply(Tracked.scala:60)
	at sbt.Classpaths$.cachedUpdate(Defaults.scala:1419)
	at sbt.Classpaths$$anonfun$updateTask$1.apply(Defaults.scala:1348)
	at sbt.Classpaths$$anonfun$updateTask$1.apply(Defaults.scala:1310)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:226)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:226)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:235)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:226)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:226)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
[error] ({git://github.com/ucb-bar/chisel-testers.git}chisel-testers/*:update) sbt.ResolveException: unresolved dependency: edu.berkeley.cs#chisel3_2.11;3.1-SNAPSHOT: not found
[error] unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found
[error] unresolved dependency: edu.berkeley.cs#firrtl-interpreter_2.11;1.1-SNAPSHOT: not found
[error] Total time: 4 s, completed Feb 1, 2017 4:34:04 PM
icluster1 [1089] /scratch/lab-templates/lab2 # vim build.sbt

ucbjrl, 2017-02-02:
If you clone a master branch, it assumes you'll want the current master
branch from all dependent branches. We currently don't publish those to
Sonatype/Maven, so you need to clone those repos and invoke:

    % sbt publish-local

on each of them yourself.

Working on one master branch assumes you're working on all. We try to keep
these in sync, but some skew invariably creeps in. If you intend to work on
any master branch, you should monitor all the chisel3 repos (chisel3
<https://github.com/ucb-bar/chisel3>, firrtl
<https://github.com/ucb-bar/firrtl>, chisel-testers
<https://github.com/ucb-bar/chisel-testers>, and firrtl-interpreter
<https://github.com/ucb-bar/firrtl-interpreter>)

jcmartin, 2017-02-02:
Unfortunately, it isn't entirely working.

Output attached.

[output.txt](https://github.com/ucb-bar/chisel-testers/files/746747/output.txt)

jcmartin, 2017-02-02:
Looking into this more, I think this is something more symptomatic of Chisel generating something incorrectly.

jcmartin, 2017-02-02:
I fixed the issue with the output generating incorrectly. Chisel 2 called the clock input "clk" in Chisel 3 it is "clock". However, I am still seeing the same issues.

One problem is that it is spitting out a stack trace 100s of times. Another issue seems to be with initialization and reset. Lastly, it is spitting out a vcd file, so I am unable to debug this further.

New output is attached.

[output.txt](https://github.com/ucb-bar/chisel-testers/files/747237/output.txt)

ucbjrl, 2017-02-02:
Please provide details on how to duplicate this.

jcmartin, 2017-02-02:
I can provide the binary executable that should run on any CentOS inst machine (icluster1,icluster2,hpse-{9,10,11,12{).

Otherwise, the process is this:
1) Check out lab-templates from the cs250 repo with the submodules.
2) Change build.sbt to use the master branches of chisel repositories.
3) publish-local the necessary repos.
4) make test-post-syn
5) Output is in build/vlsi/vcs-syn-report

ucbjrl, 2017-02-02:
Looks like I'm missing some of the synopsis environment on the millennium machines:

```
...
cd build/vlsi/vcs-sim-gl-syn && make CONFIG=DefaultConfig
make[1]: Entering directory `/nscratch/ucbjrl/cs250/lab-templates/lab2/build/vls
i/vcs-sim-gl-syn'
make[1]: *** No rule to make target `/ecad/stdcells/synopsys-32nm/typical_rvt/ve
rilog/cells.v', needed by `simv'.  Stop.
make[1]: Leaving directory `/nscratch/ucbjrl/cs250/lab-templates/lab2/build/vlsi
/vcs-sim-gl-syn'
```

I'll see if we have a substitute.

ucbjrl, 2017-02-03:
My analysis is check_addr(test_infrastructure.scala:452) is throwing an assert:
```MemError: Address 1fff8 out of memory range [0,10000)```
Possibly concurrent with this, the simulation is exiting, throwing:
```chisel3.iotesters.TestApplicationException: test application exit - exit code 0```
The latter isn't correctly handled (something is keeping it from percolating up the stack and terminating the Scala application), so it is re-asserted on every attempt to communicate with the simulation.

I'll look into the failure of the TestApplicationException to force an exit.

jcmartin, 2017-02-03:
There is something troubling that the pre-synthesis simulation works, and the post-synthesis simulation does not if that is the only problem.

ucbjrl, 2017-02-03:
The fact that TestApplicationException doesn't percolate up the stack is a "feature". It gets converted into a test failure for which upper loops don't bother to check.

ucbjrl, 2017-02-03:
Can we determine where the address value 1fff8 could be generated?

jcmartin, 2017-02-03:
@ucbjrl I hope you are being sarcastic. :)

It should be not be able to be generated in the hardware if everything is properly reset and initialized. If I have a vcd file, I can attempt to debug this issue.

chick, 2017-02-07:
@jcmartin sorry but this got passed over.  
First can I confirm that the original problem here, the directory issue and the test command issue have been solved?
Assuming they are:  In comment above you said if you have a vcd file you could debug, is there a barrier to getting a vcd file. 
I am trying now to build and test myself.  I was not able to connect to hpse-9.eecs.berkeley.edu.  I get connection refused, so I am trying on the Aspire Millennium machines.  Sounds like we might need to sit down and go over this to figure it out.  I am fairly open this afternoon.  Tomorrow I am available after 1pm.  Also available thursday morning

jcmartin, 2017-02-07:
Yes, the directory issue is solved. It isn't clear whether it is the test command, or it is some other issue.

Right now, the vcd file is not being generated, so I am unable to debug it.

@chick : 
I am free this afternoon after 2pm in Soda.

if you email your ssh public key, I can set you up with an account that will let you access the machines.

chick, 2017-02-07:
@jcmartin sounds good, see you at 2 today

Chick Markley
Staff Programmer -- Aspire Lab
University of California, Berkeley
583-2 Soda Hall, Berkeley, CA 94720-1776
chick@berkeley.edu