freechipsproject/chisel-testers #99: Verilator concurrency issue

shunshou, 2017-02-01:
Problem seems to be caused by this line:

```
   optionsManager.chiselOptions = optionsManager.chiselOptions.copy(
       runFirrtlCompiler = false)
```

What does this do??? If I remove this line, my tests pass. 

Original e-mail I sent to @chick:

---

I pulled in a bunch of changes and tried running sbt test in dsptools, but now I'm getting a really weird bug that I can't for the life of me figure out.

Basically, all of the branches, etc. I'm using are at @ https://github.com/ucb-art/Chisel3DSPDependencies

In my case, when I run sbt-test, I get 2 errors. 1 on ParameterizedAdder (which should fail b/c I removed Saturating) and 2 on my TBSpec. Jenny, who's also using this repo, gets something like 4-5 failures...

I think it has something to do with concurrency issues? But I just can't pinpoint what's going on.

If I run sbt "test-only SimpleTB.SimpleTBSpec"

It fails on two tests:

- should properly read lits with gen = fixed and expect tolerance set to 1 bit (even with finite fractional bits) *** FAILED ***
[info]   false was not true (SimpleTBwGenTypeOption.scala:361)

and 

- should properly poke/peek io delayed 1 cycle with gen = fixed and expect tolerance set to 1 bit *** FAILED ***
[info]   false was not true (SimpleTBwGenTypeOption.scala:383)

This is super weird b/c it only fails on gen types (stuff I pass in as type parameters), and only when gen = some FixedPoint. Even weirder is the fact that the test after the last mentioned one doesn't fail, even though I still use gen = fixed. In that case, the only difference is that I've removed DspReal from the module. And weirder? Is that when I comment out all of the other tests except these two and rerun, things pass...

Of course, this is only an issue with Verilator... but this is driving me crazy. Any idea what's up?

chick, 2017-02-01:
@shunshou I'm pretty surprised that removing that line fixes things. All that says is don't run firrtl as part of the chisel.Driver.execute.  I'll look over your changes more carefully right now

shunshou, 2017-02-01:
@chick see my latest PR. It's not that line. I was wrong :\ changed too many things at once...

chick, 2017-02-01:
@shunshou The problem is definitely caused by chisel-testers using the same directory for different tests, the number of files created when using the verilator backend seems to increase the likelihood of the error.  I have created a branch in dsptools that fixes the problem with a work-around that overrides the directory.  This is [fix-simpletb-dir-stomping](https://github.com/ucb-bar/dsptools/tree/fix-simpletb-dir-stomping). I am not suggesting this is the final fix, but it is a short term workaround.  The issues as I see them are:
- Two tests of the same dut get assigned the same dir by chisel-testers
 - destructive stomping is more likely with verilator, but could happen with any backend
 - it's test_run_dir/dutname<hashcode>
- One fix might be to use timestamp instead of hashcode
 - this fixes the cross stomping
 - but leads to a new directory for each test every time it is run
- chisel-testers could some how automate the clearing of the directories after each test
 - would need an easy override of this, because debugging often requires looking at these dirs
- another fix might be to default to serial execution of tests

shunshou, 2017-02-03:
@chick i don't think you ever pushed anything new to that branch. It'd prob be really convenient, if, at least as a temporary patch, this gets merged in with master...

chick, 2017-02-03:
Sorry, not sure how that happened.  It's pushed now