freechipsproject/chisel-testers #122: chisel-testers master fails to build with firrtl master

edwardcwang, 2017-03-10:
chisel-testers revision d4f29ced98790b088be6d237341c82945ee33808
firrtl revision https://github.com/ucb-bar/firrtl/commit/e571ef88f7f69b2374fa9ba86e219523645213c6

After running sbt publish-local on firrtl, chisel3, firrtl-interpreter in that order, trying to build chisel-testers with "sbt publish-local" results in the following error:

```
[info] Set current project to Chisel.iotesters (in build file:/nscratch/edwardw/chisel-testers/)
[info] Compiling 19 Scala sources to /nscratch/edwardw/chisel-testers/target/scala-2.11/classes...
[info] Main Scala API documentation to /nscratch/edwardw/chisel-testers/target/scala-2.11/api...
[info] :: delivering :: edu.berkeley.cs#chisel-iotesters_2.11;1.2-SNAPSHOT :: 1.2-SNAPSHOT :: integration :: Fri Mar 10 02:54:52 PST 2017
[info] 	delivering ivy file to /nscratch/edwardw/chisel-testers/target/scala-2.11/ivy-1.2-SNAPSHOT.xml
[info] Wrote /nscratch/edwardw/chisel-testers/target/scala-2.11/chisel-iotesters_2.11-1.2-SNAPSHOT.pom
[error] /nscratch/edwardw/chisel-testers/src/main/scala/chisel3/iotesters/ChiselMain.scala:112: class FirrtlEmitter is abstract; cannot be instantiated
[error]         (new firrtl.FirrtlEmitter).emit(firrtl.CircuitState(chirrtl, firrtl.ChirrtlForm), writer)
[error]          ^
[error] /nscratch/edwardw/chisel-testers/src/main/scala/chisel3/iotesters/ChiselMain.scala:112: class FirrtlEmitter is abstract; cannot be instantiated
[error]         (new firrtl.FirrtlEmitter).emit(firrtl.CircuitState(chirrtl, firrtl.ChirrtlForm), writer)
[error]          ^
[error] one error found
```

edwardcwang, 2017-03-10:
Replacing FirrtlEmitter with HighFirrtlEmitter allows the compile to proceed, but further errors occur:

```
[error] /nscratch/edwardw/chisel-testers/src/main/scala/chisel3/iotesters/VerilatorBackend.scala:44: class GenVerilatorCppHarness needs to be abstract, since:
[error] it has 3 unimplemented members.
[error] /** As seen from class GenVerilatorCppHarness, the missing signatures are as follows.
[error]  *  For convenience, these are usable as stub implementations.
[error]  */
[error]   def execute(state: firrtl.CircuitState): firrtl.CircuitState = ???
[error]   def inputForm: firrtl.CircuitForm = ???
[error]   def outputForm: firrtl.CircuitForm = ???
[error] class GenVerilatorCppHarness(
[error]       ^
```

edcote, 2017-03-12:
+1 I've also observed the issue.

edcote, 2017-03-15:
Thanks @edwardcwang @chick the issue appears to be resolved (at least on my end).

closed 2017-03-19 by edwardcwang