freechipsproject/chisel-testers #124: Firrtl vs. Verilator discrepancy when testing DUT with Decoupled IO's

pingfanm, 2017-03-10:
Example DUT:
```
package example

import chisel3._
import chisel3.util._

class ExampleModule extends Module {
  val io = IO(new Bundle {
    val in  = Flipped(Decoupled(Bits(16.W)))
    val out = Decoupled(Bits(16.W))
  })
  
  val delay_value = 5.U
  
  val busy = Reg(init = false.B)
  val in_reg   = Reg(init=0.U(16.W))
  io.in.ready := !busy
  
  when(io.in.valid && !busy) {
    in_reg := io.in.bits
    busy := true.B
  }
  
  val wait_counter = Reg(init=0.U(16.W))
  
  when(io.in.valid && !busy) {
    wait_counter := 0.U
  }
  
  when(busy) {
    when(wait_counter === delay_value) {
      io.out.bits := in_reg
    }.otherwise {
      io.out.bits := 0.U
      wait_counter := wait_counter + 1.U
    }
  }
  
  io.out.valid := (io.out.bits === in_reg) && (wait_counter === delay_value) && busy
  
  printf("From printf -- in: ready %d   valid %d   value %d  -- out:  ready %d  valid %d  value %d\n",
         io.in.ready, io.in.valid, io.in.bits,
         io.out.ready, io.out.valid, io.out.bits)
}
```

Tests:
```
package example.test

import chisel3._
import chisel3.util._
import chisel3.iotesters._
import org.scalatest.{Matchers, FlatSpec}

import example._

class ExampleTester(dut: ExampleModule) extends AdvTester(dut){
  wire_poke( dut.io.in.valid, 1)
  wire_poke( dut.io.in.bits, 20)
  
  val inreadySignal:BigInt = peek(dut.io.in.ready)
  println(s"From peek in.ready: $inreadySignal")
  step(1)
}

class ExampleTestFirrtl  extends ChiselFlatSpec{
  "Example" should "show decoupled port controls correctly" in {
    chisel3.iotesters.Driver(() => new ExampleModule,"firrtl"){ c =>
      new ExampleTester(c)
    }should be(true)
  }
}

class ExampleTestVerilator  extends ChiselFlatSpec{
  "Example" should "show decoupled port controls correctly" in {
    chisel3.iotesters.Driver(() => new ExampleModule,"verilator"){ c =>
      new ExampleTester(c)
    }should be(true)
  }
}
```

when running the test with "Verilator" backend, we get:
> From peek in.ready: 1
From printf -- in: ready 1   valid 1   value    20  -- out:  ready 0  valid 0  value     0

However, when running the test with "Firrtl" backend, we get:
> From peek in.ready: 1
From printf -- in: ready 0   valid 1   value 20  -- out:  ready 1  valid 0  value 34543

The printf result of in.ready is incorrectly with "Firrtl", and there's a discrepancy between Firrtl and Verilator for the same test.

chick, 2017-03-10:
@pingfanm Are you running release or master? 
There seem to be problems associated with either that i am trying to sort through right now.  Hopefully it will not take too long

pingfanm, 2017-03-10:
I'm running master.

chick, 2017-03-17:
@pingfanm The problem seems to be related to how the two backends start-up.  The interpreter is putting poisoned random numbers everywhere, and it looks like verilator is not doing that or there are some hidden cycles that are allowing things stabilize.  Is it possible to put some simple initialization logic at the beginning of the test.  I have considered making some options for the Interpreter to start up with zero's.  Let me know how important that change is to you.  Thanks

chick, 2017-03-17:
I added 
```scala
  reset(10)
  wire_poke(dut.io.in.valid, 0)
  wire_poke(dut.io.in.bits, 0)
  step(1)
``` 
To the beginning of the Test and the two backends seem to give the same results

closed 2017-03-17 by pingfanm