freechipsproject/chisel-testers #153: VCS backend fails on blocks with no IOs (standalone)

stevenmburns, 2017-06-08:
The code below fails to compile in VCS but works with Verilator and firrtl-interpreter.

````
package bug

import org.scalatest.{ Matchers, FlatSpec}

import chisel3._
import chisel3.iotesters._

class VCSBug extends Module {
  val io = IO(new Bundle)
}

class Tester(c:VCSBug) extends PeekPokeTester(c) {
  step(1)
}

class VCSTest extends FlatSpec with Matchers {
  behavior of "VCSBug"
  it should "work" in {
    chisel3.iotesters.Driver( () => new VCSBug, "vcs") { c =>
      new Tester( c)
    } should be ( true)
  }
}

class VerilatorTest extends FlatSpec with Matchers {
  behavior of "VCSBug"
  it should "work" in {
    chisel3.iotesters.Driver( () => new VCSBug, "verilator") { c =>
      new Tester( c)
    } should be ( true)
  }
}

class FirrtlTest extends FlatSpec with Matchers {
  behavior of "VCSBug"
  it should "work" in {
    chisel3.iotesters.Driver( () => new VCSBug, "firrtl") { c =>
      new Tester( c)
    } should be ( true)
  }
}
````
The trailing comma after the reset binding in the harness is the issue:
````
module test;
  reg clock = 1;
  reg reset = 1;
  always #`CLOCK_PERIOD clock = ~clock;
  reg vcdon = 0;
  reg [1023:0] vcdfile = 0;
  reg [1023:0] vpdfile = 0;

  /*** DUT instantiation ***/
  VCSBug VCSBug(
    .clock(clock),
    .reset(reset),
  );

````

aswaterman, 2017-06-08:
Is the problem the trailing comma after ```.reset(reset)```?

Is this actually legal according to the Verilog spec--i.e., is VCS wrong, or is Verilator too lax?

stevenmburns, 2017-06-08:
[image]

I don't see any trailing commas allowed in the Verilog 2001 module instantiation spec, unless the thing called "expression" is allowed to be empty. But if it was, then there would be three parameters to module and they are only two where to module is defined.

Fixing this should be easy---it is not as if verilator is going to fail if we remove the extra comma.

aswaterman, 2017-06-08:
Yeah, of course.  I asked because I was going to file a ticket if it was a VCS bug.

stevobailey, 2017-11-22:
I think it's because verilator and FIRRTL don't generate a Verilog test harness. Only VCS does.