freechipsproject/chisel-testers #155: New option "--no-check-comb-loops" doesn't work with PeekPokeTester (possibly others)

stevenmburns, 2017-06-16:
The new option to "firrtl" isn't enabled in the testing flow.
This code still flags a combinational loop. (Does work from "firrtl" (command line) and the chisel3.Driver methods.)

````
package bug

import org.scalatest.{ Matchers, FlatSpec, GivenWhenThen}

import chisel3._
import chisel3.util._
import chisel3.iotesters._

class HasCycle extends Module {
  val io = IO( new Bundle {
    val a = Input(Bool())
    val o = Output(Bool())
  })

  val b = Wire(Bool())
  b := b&&io.a

  io.o := b
}

class HasCycleTester( c:HasCycle) extends PeekPokeTester(c) {
  poke( c.io.a, 0)
  step(1)
}

class HasCycleTest extends FlatSpec with Matchers {
  behavior of "HasCycle"
  it should "work" in {
    chisel3.iotesters.Driver.execute( Array( "--no-check-comb-loops", "--backend-name", "verilator"), () => new HasCycle) { c =>
      new HasCycleTester( c)
    } should be ( true)
  }
}
````
Here is the error:
````
[info] HasCycleTest:
[info] HasCycle
[info] [0.003] Elaborating design...
[info] [0.149] Done elaborating.
[info] - should work *** FAILED ***
[info]   firrtl.transforms.CheckCombLoops$CombLoopException: : [module HasCycle] Combinational loop detected:
[info] HasCycle._T_5
[info] HasCycle.b
[info] HasCycle._T_5
[info]   at firrtl.transforms.CheckCombLoops$$anonfun$run$1$$anonfun$apply$8.apply(CheckCombLoops.scala:216)
[info]   at firrtl.transforms.CheckCombLoops$$anonfun$run$1$$anonfun$apply$8.apply(CheckCombLoops.scala:211)
[info]   at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[info]   at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[info]   at firrtl.transforms.CheckCombLoops$$anonfun$run$1.apply(CheckCombLoops.scala:211)
[info]   at firrtl.transforms.CheckCombLoops$$anonfun$run$1.apply(CheckCombLoops.scala:204)
[info]   at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[info]   at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[info]   at firrtl.transforms.CheckCombLoops.run(CheckCombLoops.scala:204)
[info]   at firrtl.transforms.CheckCombLoops.execute(CheckCombLoops.scala:231)
[info]   ...
[info] ScalaTest
````

chick, 2017-06-30:
@stevenmburns I have some PR's out to fix this, it will probably take a few days to get them reviewed and merged.  Do you have a more filled out example of a test that illustrates the behavior of these circuits that contain apparent loops.  When I put the circuit through to the interpreter, it was also flagged by the interpreters execution engine of having a loop.  There's a command line flag that let's it continue on, but I'd like to see that it handles the circuit correctly when this flag is used.

stevenmburns, 2017-06-30:
I have a firrtl pass to remove the combinational loop. Without this fix (or something like it) I can’t run it because the comb loop checker runs before my pass.
I can give you a circuit that should be simulate-able even though there are loops. I’ll gather that in a bit.