freechipsproject/chisel-testers #166: PeekPokeTester.peekAt/pokeAt broken?

jkorinth, 2017-08-25:
I've declared a memory via `val mem = Mem(sz, UInt(64.W))`, but `peekAt` just returns random values, even when using `pokeAt` on the same addresses a few cycles earlier. Is this bug, or am I missing something here? (I'm aware that peek and poke now only work on values in the top-level `IO`; but since `Mem`s cannot be declared in the Bundle, how can they be accessed efficiently in simulation?)

chick, 2017-08-25:
@jkorinth can you provide a more detailed example, please?

jkorinth, 2017-08-26:
Sure, I've created an example project at [jkorinth/peekpook.git](https://github.com/jkorinth/peekpook.git). `sbt test` illustrates the problem and/or my lack of Chisel skills. 😉 Btw: `PeekPokeTester` does not work with `SyncReadMem` at the moment?

jkorinth, 2017-09-18:
Any news on this one?

chick, 2017-09-18:
@jkorinth Yikes, somehow your example update slipped by my sensors.  Short story is nothing happened here, until today.  I am preparing a fix for the interpreter, which will make your example work.  I should have a PR out today for the firrtl-interpreter and the chisel-testers.  I have not been able to figure out yet how to fix this with verilator.  I am much less familiar with it's internals, but I'm taking a look.

chick, 2017-09-20:
@jkorinth I've taken a look at fixing this in verilator, and while I think it seems like a reasonable idea, it's going to be a while before this will get done, it is a lot more complicated to fix in verilator than it is in the interpreter

jkorinth, 2017-09-21:
@chick Great, thanks! 👍 Verilator is not so important to me as I'm running most of my tests in the interpreter anyway. Nice to have for the future, but not low priority, I'd say.
Is the fix already available in the SNAPSHOT builds available via ivy/sbt, or do I need to build Chisel from source?

chick, 2017-09-21:
The change is not yet available in snapshot release, it should be out by next week.  In the meantime you would need to build chisel-testers and firrtl-interpreter from source, and it's very probable that there are dependencies on firrtl and chisel as well.

jkorinth, 2017-09-22:
Just built from source (surprisingly easy, 👍) and can confirm the issue is now fixed. Thanks @chick!

closed 2017-09-22 by jkorinth