freechipsproject/chisel-testers #179: Issues with println

jackkoenig, 2017-11-08:
It appears that println is changed to instead call logger.info. I think this is kind of problematic because it is unexpected and it also accepts String rather than Any so you have to explicitly call .toString on arguments to it.

I think we should just mess around with println and encourage people to use the logger when they need multiple levels of verbosity.

chick, 2017-11-08:
@jackkoenig 
I agree, not even sure how it happened.
So should I just remove that statement.
All tests pass when it is removed.
I don't think this is an API breaking change since *println*'s signature is more permissive

jackkoenig, 2017-11-08:
Yeah I agree, just removing the statement is the right move.