freechipsproject/chisel-testers #183: Ability to use PeekPokeTester similar to InterpretiveTester

wachag, 2018-01-15:
It is a bit complex to phrase what I want, but I'll try it nonetheless
PeekPokeTester requires that every test should be in the tester class, like:

class GCDPeekPokeTester(c: RealGCD2) extends PeekPokeTester(c)  {
    poke(c.io.in.bits.a, 1)
}

With InterpretiveTester one has more degree of freedom:

val tester = new InterpretiveTester(s)
tester.poke("io_a", i)

Note: in case of the InterpretiveTester one can create a class, which instantiates an InterpretiveTester, and has a poke() and a peek() method, thus allowing more freedom. This use case can happen in complex verification environments, for example when you want to mimic UVM. An example could be this one (this could be done inside a PeekPokeTester class, but when the sequencers get more complex or you want to do distributed simulation it could be not):

class DUTTester {
  def poke(...)=...
  def peek(...)=...
 ... initialization...
}

object Verif extends App{
  val seq=new Sequencer()
  val dut=new DutTester()
 ...
 dut.poke(seq.next())
 ...
}

In most cases InterpretiveTester could do the job, however in some cases one wants to use the Verilator backend to test on Verilog level.

wachag, 2018-01-15:
One can hack some in a non intended way to have an instance of the PeekPokeTester class, and could to this:
var testerClass: Tester[T] = null
  Driver.execute(gen, options) { c: T => {
    testerClass = new Tester[T](c)
    testerClass
  }
  }
Where Tester extends PeekPokeTester. Then one can use testerClass.peek() and testerClass.poke(). However this one fails to do the intented (but of course not supported) behavior because 
https://github.com/freechipsproject/chisel-testers/blob/b897bba427fe8de14e5b59a9765a673b82af0447/src/main/scala/chisel3/iotesters/PeekPokeTester.scala#L300
When the constructor of a PeekPokeTester finishes, it automatically stops the simulation.

chick, 2018-01-15:
I'm not quite sure what the question is here but I have a feeling that the following test construction example may be relevant. Take a look at [/ChiselPokeSpec.scala](https://github.com/freechipsproject/chisel-testers/blob/master/src/test/scala/chisel3/iotesters/ChiselPokeSpec.scala), it separates out the tester from the DUT creation in a different way and should be runnable with verilog.  Let me know if I am not on the right track here.