freechipsproject/chisel-testers #220: OrderedDecoupledHWIOTester only checks only as many outputs as there are inputs

nhynes, 2018-11-17:
### Details

**Type of issue**: bug report

**Impact**: API modification

**Description**: `OrderedDecoupledHWIOTester` does not process more output events than there are input events. This prevents testing circuits which produce multiple outputs for a single input (e.g., bit-width converters).

### Minimum working example

```scala
import chisel3._
import chisel3.util._
import chisel3.iotesters.{ChiselFlatSpec, Driver, OrderedDecoupledHWIOTester, PeekPokeTester}

class ParallelToSerialTester extends OrderedDecoupledHWIOTester {
  val MaxCount = 4 // setting `MaxCount = 1` works fine

  val device_under_test = Module(new Module {
    val io = IO(new Bundle {
      val in = Flipped(Decoupled(Bool()))
      val out = Decoupled(UInt(MaxCount.U.getWidth.W))
    })

    io.in.ready := true.B
    io.out.valid := true.B
    val (ctr, _ding) = Counter(true.B, MaxCount)
    io.out.bits := ctr
  })

  inputEvent(device_under_test.io.in.bits -> 1)
  for (i <- 0 until MaxCount)
    outputEvent(device_under_test.io.out.bits -> i)
}
```

### Output from tester
```
output test event  0 testing out.bits =  0, should be  0
Exceeded maximum allowed 10000 ticks in OrderedDecoupledHWIOTester,
If you think code is correct use:
DecoupleTester.max_tick_count = <some-higher-value>
```

### Expected behavior
```
output test event  0 testing out.bits =  0, should be  0
output test event  1 testing out.bits =  1, should be  1
output test event  2 testing out.bits =  2, should be  2
output test event  3 testing out.bits =  3, should be  3
```

chick, 2018-11-19:
I will try and take a look at this over the next week or two, but at the moment, the SteppedHardwareIOTester is not being actively worked on or supported.  We are working on moving to a new paradigm for the PeekPokeTesters. See [https://github.com/ucb-bar/chisel-testers2](https://github.com/ucb-bar/chisel-testers2) and [https://github.com/freechipsproject/chisel3/issues/725](https://github.com/freechipsproject/chisel3/issues/725). Is there a particular use case where the Stepped tester better meets those requirements?

nhynes, 2018-11-19:
Woah, I saw the PPT proposal issue but didn't know that there was a `chisel-testers2`! I'll give that a shot. Thanks!

closed 2018-11-19 by nhynes