freechipsproject/chisel-testers #230: Parameterized output simulation using VCS

sijundu, 2019-01-11:
There is a boolean parameter named `debugIO` in my code and it is used to choose if using a series of outputs according to the value of this parameter. A part of the IO bundle is shown as:

```
  val globalCounter  = if(p.debugIO){Output(UInt(p.widthGlobalCounter.W))} else{Output(UInt(0.W))}
  val state          = if(p.debugIO){Output(UInt(4.W))} else{Output(UInt(0.W))}
  val noiseAve       = if(p.debugIO){Output(UInt(p.widthCorr.W))} else{Output(UInt(0.W))}
  val snrValue       = if(p.debugIO){Output(UInt(10.W))} else{Output(UInt(0.W))}
  val max            = if(p.debugIO){Output(UInt(p.widthCorr.W))} else{Output(UInt(0.W))}
  val maxIndex       = if(p.debugIO){Output(UInt(p.widthPhase.W))} else{Output(UInt(0.W))}
  val max2           = if(p.debugIO){Output(UInt(p.widthCorr.W))} else{Output(UInt(0.W))}
  val max2Index      = if(p.debugIO){Output(UInt(p.widthPhase.W))} else{Output(UInt(0.W))}
  val regCACount     = if(p.debugIO){Output(UInt(p.widthPhase.W))} else{Output(UInt(0.W))}
  val regIntegCount  = if(p.debugIO){Output(UInt(p.widthCorrSample.W))} else{Output(UInt(0.W))}
  val regNbInteg     = if(p.debugIO){Output(UInt((log2Ceil(p.nInteg)+1).W))} else{Output(UInt(0.W))}
  val regPhaseIndex  = if(p.debugIO){Output(UInt(p.widthPhase.W))} else{Output(UInt(0.W))}
  val regShiftCA     = if(p.debugIO){Output(Bool())} else{Output(SInt(0.W))}
  val sampleDelay    = if(p.debugIO){Output(UInt(p.widthGlobalCounter.W))} else{Output(UInt(0.W))}
  val cos            = if(p.debugIO){Output(SInt(p.widthNCOTrunct.W))} else{Output(SInt(0.W))}
  val sin            = if(p.debugIO){Output(SInt(p.widthNCOTrunct.W))} else{Output(SInt(0.W))}
  val regSum_i       = if(p.debugIO){Output(SInt(p.widthSumIQ.W))} else{Output(SInt(0.W))}
  val regSum_q       = if(p.debugIO){Output(SInt(p.widthSumIQ.W))} else{Output(SInt(0.W))}
  val sum            = if(p.debugIO){Output(UInt(p.widthSumCorr.W))} else{Output(UInt(0.W))}
  val caOut          = if(p.debugIO){Output(Bool())} else{Output(UInt(0.W))}
  val wireShiftCA    = if(p.debugIO){Output(SInt(2.W))} else{Output(SInt(0.W))}
```

When I set `p.debugIO` as `true`, everything works well. But when I set it to `false`, the simulation fails with the following error message:

```
[info] Loading project definition from /tools/B/sijun/projects/GPS/github/sijun/project
[info] Loading settings for project sijun from build.sbt ...
[info] Set current project to gps (in build file:/tools/B/sijun/projects/GPS/github/sijun/)
[info] [0.003] Elaborating design...
[info] [0.572] Done elaborating.
Total FIRRTL Compile Time: 21487.9 ms
cd test_run_dir/gps.AcquisitionChannelTester1129716188 && vcs -full64 -quiet -timescale=1ns/1ps -debug_pp -Mdir=AcquisitionChannelWrapper.csrc +v2k +vpi +vcs+lic+wait +vcs+initreg+random +define+CLOCK_PERIOD=1 -P vpi.tab -cpp g++ -O2 -LDFLAGS -lstdc++ -CFLAGS "-I$VCS_HOME/include -I$dir -fPIC -std=c++11" -o AcquisitionChannelWrapper AcquisitionChannelWrapper.v AcquisitionChannelWrapper-harness.v vpi.cpp

Warning-[IPDW] Identifier previously declared
AcquisitionChannelWrapper-harness.v, 4
  Second declaration for identifier 'clock' ignored
  Identifier 'clock' previously declared as reg. 
  [AcquisitionChannelWrapper-harness.v, 2]

Warning-[IPDW] Identifier previously declared
AcquisitionChannelWrapper-harness.v, 7
  Second declaration for identifier 'reset' ignored
  Identifier 'reset' previously declared as reg. 
  [AcquisitionChannelWrapper-harness.v, 3]

Warning-[DPIMI] Duplicate port in module instantiation
AcquisitionChannelWrapper-harness.v, 103
"AcquisitionChannelWrapper AcquisitionChannelWrapper( .clock (clock),  .reset (reset),  .clock (clock_delay),  .reset (reset_delay),  .io_integTime (io_integTime_delay),  .io_snrThresh (io_snrThresh_delay),  .io_svID (io_svID_delay),  .io_dataADC (io_dataADC_delay),  .io_regShiftCA (io_regShiftCA_delay),  .io_wireShiftCA (io_wireShiftCA_delay),  .io_sum (io_sum_delay),  .io_regSum_q (io_regSum_q_delay),  .io_regSum_i (io_regSum_i_delay),  .io_sin (io_sin_delay),  .io_cos (io_cos_delay),  .io_sampleDelay (io_sampleDelay_delay),  .io_regPhaseIndex (io_regPhaseIndex_delay),  .io_regNbInteg (io_regNbInteg_delay),  .io_regIntegCount (io_regIntegCount_delay),  .io_regCACount (io_regCACount_delay),  .io_caOut (io_caOut_delay),  .io_max2Index (io_max2Index_delay ... "
  Port 'clock' is connected more than once for instance 
  'AcquisitionChannelWrapper' of 'AcquisitionChannelWrapper'.
  Extra connection will be ignored.

Warning-[DPIMI] Duplicate port in module instantiation
AcquisitionChannelWrapper-harness.v, 103
"AcquisitionChannelWrapper AcquisitionChannelWrapper( .reset (reset),  .clock (clock_delay),  .reset (reset_delay),  .io_integTime (io_integTime_delay),  .io_snrThresh (io_snrThresh_delay),  .io_svID (io_svID_delay),  .io_dataADC (io_dataADC_delay),  .io_regShiftCA (io_regShiftCA_delay),  .io_wireShiftCA (io_wireShiftCA_delay),  .io_sum (io_sum_delay),  .io_regSum_q (io_regSum_q_delay),  .io_regSum_i (io_regSum_i_delay),  .io_sin (io_sin_delay),  .io_cos (io_cos_delay),  .io_sampleDelay (io_sampleDelay_delay),  .io_regPhaseIndex (io_regPhaseIndex_delay),  .io_regNbInteg (io_regNbInteg_delay),  .io_regIntegCount (io_regIntegCount_delay),  .io_regCACount (io_regCACount_delay),  .io_caOut (io_caOut_delay),  .io_max2Index (io_max2Index_delay),  .io_max2 (io_ ... "
  Port 'reset' is connected more than once for instance 
  'AcquisitionChannelWrapper' of 'AcquisitionChannelWrapper'.
  Extra connection will be ignored.

Error-[UPIMI-E] Undefined port in module instantiation
AcquisitionChannelWrapper-harness.v, 103
  Port "io_regShiftCA" is not defined in module 'AcquisitionChannelWrapper' 
  defined in "AcquisitionChannelWrapper.v", 208111
  Module instance: AcquisitionChannelWrapper AcquisitionChannelWrapper( .clock
  (clock_delay),  .reset (reset_delay),  .io_integTime (io_integTime_delay),  
  .io_snrThresh (io_snrT ...
[... 83 lines cut ...]
[info]   at scala.Predef$.assert(Predef.scala:204)
[info]   at chisel3.iotesters.setupVCSBackend$.apply(VCSBackend.scala:175)
[info]   at chisel3.iotesters.Driver$.$anonfun$execute$2(Driver.scala:60)
[info]   at scala.runtime.java8.JFunction0$mcZ$sp.apply(JFunction0$mcZ$sp.java:12)
[info]   at logger.Logger$.$anonfun$makeScope$2(Logger.scala:138)
[info]   at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[info]   at logger.Logger$.makeScope(Logger.scala:136)
[info]   at chisel3.iotesters.Driver$.$anonfun$execute$1(Driver.scala:39)
[info]   at scala.runtime.java8.JFunction0$mcZ$sp.apply(JFunction0$mcZ$sp.java:12)
[info]   at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[info]   ...
[info] ScalaTest
[info] Run completed in 26 seconds, 663 milliseconds.
[info] Total number of tests run: 1
[info] Suites: completed 1, aborted 0
[info] Tests: succeeded 0, failed 1, canceled 0, ignored 0, pending 0
[info] *** 1 TEST FAILED ***
[error] Failed: Total 1, Failed 1, Errors 0, Passed 0
[error] Failed tests:
[error]         gps.AcquisitionChannelSpec
[error] (Test / testOnly) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 29 s, completed Jan 10, 2019 3:31:59 PM
```

The error message was shown when simulating with VCS. But when I change to simulate with Verilator, the errors disappear and simulation goes normally.

edwardcwang, 2019-01-11:
A somewhat different approach: have you tried the following?

```scala
  val globalCounter  = if(p.debugIO){Some(Output(UInt(p.widthGlobalCounter.W)))} else None
```

sijundu, 2019-01-15:
What data type will `.get` point to? The same as the `io.globalCounter` ?

sijundu, 2019-01-15:
Thank you, Edward and Jack. The VCS simulation issue has been fixed with the use of `Some` and `.get`.

closed 2019-05-03 by ucbjrl