freechipsproject/chisel-testers #231: getWidth on a Bundle type does not return the right value in a tester(PeekPokeTester)

aayupov, 2019-01-18:
```scala
class LinkNode extends Bundle {
  val next = UInt(64.W)
  val count = UInt(32.W)
  override def cloneType = (new LinkNode).asInstanceOf[this.type]
}
```

See code below for which simulation log reports:
In Bla 96
[info] [0.001] In Foo 32

------------------------

```scala
class Foo (bla: Bla) extends PeekPokeTester(bla) {
  println(s"In Foo ${(new LinkNode).getWidth}")
}

class Bla extends Module {
  val io = IO (new Bundle())
  println(s"In Bla ${(new LinkNode).getWidth}")
}

class SorterAccTester extends ChiselFlatSpec {
  "FooBla" should "simulate" in {
    chisel3.iotesters.Driver(() => new Bla,"firrtl"){ c=>
      new Foo(c)
    }should be(true)
  }
}
```

chick, 2019-01-18:
I think is best answered in the  near term with chisel-testers2.

chick, 2019-06-29:
I just retested this using chisel-testers and it seems to work properly now.
In addition, bundle literals now work and also seem to have the correct behavior

closed 2019-06-29 by chick