freechipsproject/chisel-testers #242: Deprecated Driver since Firrtl 1.2

tampler, 2019-05-03:
Hello
I get a message from FIRRTL when using the iotesters.driver.execute, as in the chisel-template:
```bash
Warning: firrtl.Driver is deprecated since 1.2!
Please switch to firrtl.stage.FirrtlStage
```

How do I fix that ? Using latest snapshots for all tools

Thanks,
Boris

sequencer, 2019-05-03:
It might be a temporary issue since https://github.com/freechipsproject/firrtl/pull/1005 has been merged, while https://github.com/freechipsproject/chisel3/pull/1004 hasn’t.
Maybe need to wait https://github.com/freechipsproject/chisel3/pull/1004 to finish the compatible layer for chisel3.Driver

seldridge, 2019-05-03:
`firrtl.Driver` is deprecated since Chisel 3.2 / FIRRTL 1.2 and is expected to be removed in Chisel 3.3 / FIRRTL 1.3. This is part of the broader Stage/Phase refactor where all drivers will be deprecated in favor of new stages. The process for this is that each `Driver` is deprecated and becomes a compatibility wrapper around an equivalent stage, e.g., `firrtl.Driver` is deprecated in favor of `firrtl.stage.FirrtlStage` and `chisel3.Driver` is deprecated in favor of `chisel3.stage.ChiselStage`. While these changes percolate through all the projects, any project that calls a driver that has not been converted to a compatibility layer will get this deprecation warning. This is currently the situation for `chisel3.Driver` which still calls `firrtl.Driver` on the master branch.

For your case, @tampler, there's nothing you can do. :cry: As @sequencer mentions, I think this will be cleared up once `ChiselStage` lands. The `ChiselStage` PR will call `FirrtlStage` internally and remove that deprecation warning.

If this is something that you want to dig into in more detail, the PRs that @sequencer references provide a fair amount of information. The overall intent, after much gnashing of teeth, was to try to preserver backwards compatibility via the drivers so that we can merge the PRs one at a time. My earlier approach to this would have required simultaneous bumps to every project. :scream:

Apologies for blowing up your terminal with more messages. These deprecation warnings should be gone once 3.2/1.2 is actually released.

tampler, 2019-05-03:
Thanks everybody for a clear response. Waiting for the next Chisel release

closed 2019-05-03 by tampler