freechipsproject/chisel-testers #246: Supporting ZIO fibers

tampler, 2019-05-04:
I'm working to integrate [ZIO Fibers](https://scalaz.github.io/scalaz-zio/) in my Chisel simulations.
Can't make it work now, inviting developers into this [discussion](https://github.com/scalaz/scalaz-zio/issues/821).

Your ideas are highly welcome

Best,
Boris

chick, 2019-05-06:
I will take a look at this starting tomorrow afternoon, I'm in the middle of something I have to finish and I won't be free until then. Sorry for the delay, but I will get to it as soon as I can

chick, 2019-05-08:
I think your problems are because the step method called in your parallel routines are not linked.
The simulator sees the steps as separate and you get sequential events in your waveforms instead of parallel events. 
While I will continue to work to look at this. In the meantime I would strongly suggest that you take a look at [testers2](https://github.com/ucb-bar/chisel-testers2), it builds in a parallel model with strong synchronization semantics. Testers2 should be come the default unit test framework and there has been a lot of improvements over chisel-testers. If it is not the answer it may still have clues that help us solve this.
Right now testers2 only supports the treadle backend, but one of my priority todo items is to add the other backends to testers2, if the testers2 model looks like it will work and you need verilator I can up the priority getting that item done.
In the meantime I will see if I can get your example from the fibers thread mocked up (it would help if you have a simple mock of your DUT already).

chick, 2019-05-08:
This is my mockup DUT
```scala
import chisel3._
import chisel3.core.MultiIOModule
import chisel3.util.Decoupled

class TpuInput extends Bundle {
  val data = UInt(64.W)
}

class TpuMockup extends MultiIOModule {
  val adin = IO(Vec(2, Flipped(Decoupled(new TpuInput))))
  val out  = IO(Output(Vec(2, UInt(64.W))))

  adin := DontCare
  out  := DontCare

  val reg = RegInit(VecInit(Seq.fill(2)(0.U(64.W))))

  for(index <- adin.indices) {
    when(adin(index).valid) {
      val newData = adin(index).deq()
      reg(index) := reg(index) +% newData.data
      out(index) := reg(index)
      printf("adding %d to reg who's current value is %d\n", newData.data, reg(index))
    }
  }
}
```

chick, 2019-05-08:
Here is a testers2 driver that uses two parallel blocks, even though they are run independently testers two understands that they share a clock so input events are triggered properly. The tests adds the 16 numbers from the two for blocks in 16 cycles
```scala
import chisel3._
import chisel3.experimental.BundleLiterals._
import chisel3.iotesters.{TesterOptions, TesterOptionsManager}
import chisel3.tester._
import org.scalatest.FreeSpec

class TpuMockupSpec extends FreeSpec with ChiselScalatestTester {
  "parallel events should work" in {
    val manager = new TesterOptionsManager {
      testerOptions = testerOptions.copy(generateVcdOutput = "on")
      treadleOptions = treadleOptions.copy(writeVCD = true)
    }
    (new TestBuilder(() => new TpuMockup, Some(manager), None)) { dut =>

      dut.adin(0).initSource()
      dut.adin(0).setSourceClock(dut.clock)
      dut.adin(1).initSource()
      dut.adin(1).setSourceClock(dut.clock)

      def pokeStuff(index: Int): Unit = {
        val inputs = for {
          i <- 0 until 4
          j <- 0 until 4
        } yield (new TpuInput).Lit(_.data -> (index * 100 + i * 10 + j).U)

        dut.adin(index).enqueueSeq(inputs)
      }

      fork {
        pokeStuff(0)
      }.fork {
        pokeStuff(1)
      }.join()

      dut.clock.step()

      for (index <- dut.adin.indices) {
        println(s"final value of reg($index} is ${dut.out(index).peek()}")
      }
    }
  }
}
```

chick, 2019-05-08:
```
[info] [0.005] Elaborating design...
[info] [1.471] Done elaborating.
------------------------------------------------------------------------------
Warning: firrtl.Driver is deprecated since 1.2!
Please switch to firrtl.stage.FirrtlStage
------------------------------------------------------------------------------
Total FIRRTL Compile Time: 739.3 ms
Total FIRRTL Compile Time: 294.4 ms
file loaded in 0.625099393 seconds, 35 symbols, 29 statements
Clock tick 0
adding 100 to reg who's current value is 0
adding 0 to reg who's current value is 0
Clock tick 1
adding 101 to reg who's current value is 100
adding 1 to reg who's current value is 0
Clock tick 2
adding 102 to reg who's current value is 201
adding 2 to reg who's current value is 1
Clock tick 3
adding 103 to reg who's current value is 303
adding 3 to reg who's current value is 3
Clock tick 4
adding 110 to reg who's current value is 406
adding 10 to reg who's current value is 6
Clock tick 5
adding 111 to reg who's current value is 516
adding 11 to reg who's current value is 16
Clock tick 6
adding 112 to reg who's current value is 627
adding 12 to reg who's current value is 27
Clock tick 7
adding 113 to reg who's current value is 739
adding 13 to reg who's current value is 39
Clock tick 8
adding 120 to reg who's current value is 852
adding 20 to reg who's current value is 52
Clock tick 9
adding 121 to reg who's current value is 972
adding 21 to reg who's current value is 72
Clock tick 10
adding 122 to reg who's current value is 1093
adding 22 to reg who's current value is 93
Clock tick 11
adding 123 to reg who's current value is 1215
adding 23 to reg who's current value is 115
Clock tick 12
adding 130 to reg who's current value is 1338
adding 30 to reg who's current value is 138
Clock tick 13
adding 131 to reg who's current value is 1468
adding 31 to reg who's current value is 168
Clock tick 14
adding 132 to reg who's current value is 1599
adding 32 to reg who's current value is 199
Clock tick 15
adding 133 to reg who's current value is 1731
adding 33 to reg who's current value is 231
Clock tick 16
final value of reg(0} is UInt<64>(264)
final value of reg(1} is UInt<64>(1864)
Clock tick 17
```

chick, 2019-05-08:
[image]

chick, 2019-05-08:
Short story: I think ZIO or any parallel framework for scala is going to be hard to use with chisel-testers because of coordination with the single threaded backend. Testers2 is an attempt to address this difficulty. I hope this example gives you some ideas on how to approach these problems. Let me know

tampler, 2019-05-09:
Hello @chick  
I created a [new project](https://github.com/freechipsproject/twerk) for this and added all my stuff there. We may work on that test project to discover possibilities.

To run it, simply launch the sbt shell and type "test"

tampler, 2020-03-12:
Not actual anymore

closed 2020-03-12 by tampler