freechipsproject/chisel-testers #247: [BUG] verilator: 0-width signals in Decoupled Bundle lead to incorrect peek

johnsbrew, 2019-05-10:
Please find below a self-explanatory code snippet with proper tests.

I spent some time to shrink the original piece of code to isolate the bug, however I am not able to understand the root cause of this bug and even whether it lies in the PeekPokeTester or somewhere else.
So here are the facts:
- Waves are corrects (`io.res.valid` stuck to 0)
- `peek(c.io.res.valid)` always returns 1
- this behavior happens only with verilator and when the width of `extra` is 0 in the decoupled bundle `res_flow`

```scala
import chisel3._
import chisel3.util._
import chisel3.iotesters._

class SimpleBug (
  val width: Int,
  val extra_width : Int = 0
) extends Module {

  val res_flow = new Bundle {
    val data  = UInt(width.W)
    val extra = UInt(extra_width.W)
  }

  val io = IO(new Bundle {
    val req    = Flipped(Decoupled(UInt(width.W)))
    val res    = Decoupled(res_flow)
    val extra  = Flipped(Decoupled(UInt(extra_width.W)))
  })
  
  val out_req          = Reg(UInt(width.W))
  val out_valid        = RegInit(Bool(),false.B)  

  when(io.req.ready){
    out_req        := io.req.bits
    out_valid      := io.req.valid
  }
  io.req.ready     := (io.res.ready && io.res.valid) || !out_valid
  io.extra.ready    := false.B
  
  io.res.valid      := out_valid
  io.res.bits.extra := io.extra.bits
  io.res.bits.data  := out_req

}

class SimpleTester(c: SimpleBug) extends PeekPokeTester(c) {
  reset(1)
  step(1)
  poke(c.io.req.valid, false)
  poke(c.io.req.bits, 42)
  for(i <- 0 to 3) {
    expect(c.io.res.valid, false)
    step(1)
  }
  step(10)
}

class SimpleBugSpec extends ChiselFlatSpec {
  val manager = new TesterOptionsManager {
    testerOptions = testerOptions.copy(isVerbose = true)
    testerOptions = testerOptions.copy(backendName = "verilator")
  }
  "SimpleBug" should "fail with zero width field in bundle" in {
    chisel3.iotesters.Driver.execute(() => new SimpleBug(
      width = 10
    ), manager) {
      c => new SimpleTester(c)
    } should be (true)
  }
  "SimpleBug" should "work with positive width field in bundle" in {
    chisel3.iotesters.Driver.execute(() => new SimpleBug(
      width = 10,
      extra_width = 5
    ), manager) {
      c => new SimpleTester(c)
    } should be (true)
  }
}
```

Please notice that both testcases run fine with treadle backend (I did not test any other ones)

@chick any idea or workaround to suggest ?

chick, 2019-05-10:
I don't know what this is yet. Happens here for me as well.
Thanks for the good work illustrating the problem.
I'll fiddle with it today and hopefully I'll turn up something

chick, 2019-05-10:
@johnsbrew I believe this is fixed with PR #239 which was just merged a few days ago.
Seems to run properly and match Treadle for me in my test set-up.
Problem was in the SimApiInterface. Was adding in the zero length fields as non-zero
when computing some kind of index.
Let me know if this seems to work for you.

johnsbrew, 2019-05-13:
#239 does fix this bug, thank you @chick :)

closed 2019-05-13 by johnsbrew

hukangha, 2019-09-11:
I also got 0-width issue, 
with poke 
https://github.com/freechipsproject/chisel-testers/issues/258