freechipsproject/chisel-testers #258: verilator poke incorrect signal for 0-width RRArbiter

hukangha, 2019-09-11:
**Type of issue**: bug report 

**Impact**:  unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
I'm Trying to get if RRArbiter works when discarding the bits signal with **UInt(0.W)** and only use the valid and chosen signal
so I came up with this testing code
```scala
class ArbiterTester(c: RRArbiter[UInt] ) extends PeekPokeTester(c) {

  private val abtr = c
  poke(abtr.io.out.ready, 1)
  poke(abtr.io.in(0).valid, 0)
  poke(abtr.io.in(1).valid, 0)
  poke(abtr.io.in(2).valid, 1)
  poke(abtr.io.in(3).valid, 0)
  poke(abtr.io.in(4).valid, 0)
  println(s"out.chosen: ${peek(abtr.io.chosen)}")
  expect(abtr.io.out.valid, 1)
  step(1)
  poke(abtr.io.in(1).valid, 1)
  poke(abtr.io.in(3).valid, 1)
  println(s"out.chosen: ${peek(abtr.io.chosen)}")
  expect(abtr.io.out.valid, 1)
  step(1)
  println(s"out.chosen: ${peek(abtr.io.chosen)}")
  expect(abtr.io.out.valid, 1)
  step(1)
  println(s"out.chosen: ${peek(abtr.io.chosen)}")
  expect(abtr.io.out.valid, 1)
  step(10)

}

class ABTTester extends ChiselFlatSpec {
  // Disable this until we fix isCommandAvailable to swallow stderr along with stdout
  private val backendNames = if(firrtl.FileUtils.isCommandAvailable(Seq("verilator", "--version"))) {
    Array("firrtl", "verilator")
  }
  else {
    Array("firrtl")
  }

  for ( backendName <- backendNames ) {
    "arbiter" should s"get good qos (with $backendName)" in {
      Driver(() => new RRArbiter(UInt(0.W),5), backendName) {
        c => new ArbiterTester(c)
      } should be (true)
    }
  }

}
```

**What is the current behavior?**
the input signal is incorrect shown in the vcd generated by verilator:
[image]

**What is the expected behavior?**
this is the correct signal when setting the type to **UInt(1.W):**
[image]

as I can get correct verilog code when discarding the bits signal with UInt(0.W),
seems the simulation should work properly.

**Please tell us about your environment:**
```scala
version := "3.1.1"

scalaVersion := "2.11.12"

crossScalaVersions := Seq("2.11.12", "2.12.4")

resolvers ++= Seq(
  Resolver.sonatypeRepo("snapshots"),
  Resolver.sonatypeRepo("releases")
)

// Provide a managed dependency on X if -DXVersion="" is supplied on the command line.
val defaultVersions = Map(
  "chisel3" -> "3.1.+",
  "chisel-iotesters" -> "[1.2.5,1.3-SNAPSHOT["
  )

libraryDependencies ++= Seq("chisel3","chisel-iotesters").map {
  dep: String => "edu.berkeley.cs" %% dep % sys.props.getOrElse(dep + "Version", defaultVersions(dep)) }
```
    - OS: `4.15.0-58-generic ubuntu18.04`

**What is the use case for changing the behavior?**

--

hukangha, 2019-09-11:
Hi,
How should I manage the dependencies? 
I changed the version to
```scala
val defaultVersions = Map(
  "chisel3" -> "3.1.+",
  "chisel-iotesters" -> "1.3-SNAPSHOT"
  )
```
it reports:
sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#chisel3_2.12.+;3.1.+: not found

johnsbrew, 2019-09-11:
This error about chisel3 as 3.1.+ is quite weird since you didn't change the requirements for chisel and the package should already be in your ivy cache...
 
Nonetheless using chisel-iotesters as 1.3-SNAPSHOT is likely to require to also use chisel as 3.2-SNAPSHOT.
Using both snapshots is my current setup for my projects and it works well.

hukangha, 2019-09-11:
Thanks, it's working.
btw,
how do you feel about the snapshot version?
is it stable enough for foundry for ASIC ?

johnsbrew, 2019-09-11:
> btw,
> how do you feel about the snapshot version?
> is it stable enough for foundry for ASIC ?

Well, I am a FPGA guy so I cannot really help you with ASIC stability concern, maybe @jackkoenig or @azidar would have interesting inputs on that matter?

IMHO it doesn't really matter since the design you verify (with verilator) is the generated verilog, which is what you will then synthesize to your build your ASIC. 
If you are confident enough in your testbenches/test-cases I would tend to say that snapshots or not doesn't matter. Btw using a released version of chisel does not come with any additional guarantees.
I don't know what is the actual approach of people building ASIC like @jackkoenig : how much do you trust verilator ? VCS/Questa? do you use the chisel-tester ? Custom C/Verilog test-benches ?

chick, 2019-09-11:
@hukangha I am not sure of the status here. Did changing the versions fix the problem.
I ran your tests myself with current masters and I believe it ran correctly.
Does your test fail when the behavior is incorrect?

hukangha, 2019-09-16:
> @hukangha I am not sure of the status here. Did changing the versions fix the problem.
> I ran your tests myself with current masters and I believe it ran correctly.
> Does your test fail when the behavior is incorrect?

Hi,@chick
The test is OK with the 1.3-SNAPSHOT version.
just came back to work from vacation~

closed 2019-09-16 by hukangha