freechipsproject/chisel-testers #267: Can't poke MixedVec

Martoni, 2019-11-05:
I already asked the question on [stackoverflow](https://stackoverflow.com/questions/58374872/cant-poke-mixedvec) but i reproduce the issue here because it seems to be a tester bug.

I updraded the project WbPlumbing to use last version of Chisel and iotester in [build.sbt ](https://github.com/Martoni/WbPlumbing/blob/master/build.sbt):
```scala
val defaultVersions = Map(
  "chisel3" -> "3.2.+",
  "chisel-iotesters" -> "1.3.+"
)
```
But I still have error when I try to make tests : 
```
$ make test
[...]
[info] WbInterconOneMasterSpec:
[info] A WbInterconOneMaster
[info] - should read and write wishbone value on two slaves *** FAILED ***
[info]   chisel3.internal.ChiselException: Error: Not in a RawModule. Likely cause: Missed Module() wrap, bare chisel API call, or attempting to construct hardware inside a BlackBox.
[info]   at chisel3.internal.throwException$.apply(Error.scala:42)
[info]   at chisel3.internal.Builder$.referenceUserModule(Builder.scala:287)
[info]   at chisel3.Data.connect(Data.scala:384)
[info]   at chisel3.Data.$colon$eq(Data.scala:475)
[info]   at wbplumbing.TestWbInterconDualSlave.$anonfun$new$15(testwbplumbing.scala:63)
[info]   at wbplumbing.TestWbInterconDualSlave.$anonfun$new$15$adapted(testwbplumbing.scala:62)
[info]   at scala.collection.Iterator.foreach(Iterator.scala:941)
[info]   at scala.collection.Iterator.foreach$(Iterator.scala:941)
[info]   at scala.collection.AbstractIterator.foreach(Iterator.scala:1429)
[info]   at scala.collection.IterableLike.foreach(IterableLike.scala:74)
[info]   ...
[info] ScalaTest
[info] Run completed in 6 seconds, 639 milliseconds.
[info] Total number of tests run: 1
[info] Suites: completed 1, aborted 0
[info] Tests: succeeded 0, failed 1, canceled 0, ignored 0, pending 0
[info] *** 1 TEST FAILED ***

```

edwardcwang, 2019-11-05:
Hi Fabien, does it work with testers2? These testers are no longer being actively developed.

Martoni, 2019-11-05:
but testers2 is «on the bleeding edge» in alpha version. You think we should only use testers2 now ?

edwardcwang, 2019-11-06:
iotesters will continue to be 'supported' but since MixedVec is a new feature, my impression is that it's not a high priority to support it on the older testers.

schoeberl, 2019-11-06:
When is the plan that tester2 becomes the “official” Chisel 3 testing framework?

I am asking as I prepare teaching material for the spring semester.

Cheers,
Martin

Martoni, 2019-11-06:
I'm trying the tester2 but it seems to not working with [last chisel 3.2.+](https://github.com/ucb-bar/chisel-testers2/issues/72).

Martoni, 2019-11-06:
It's working, but need chisel 3.3-SNAPSHOT. Really on the bleeding edge ;)

closed 2019-11-14 by Martoni