freechipsproject/chisel-testers #271: Peek/Poke/Step/Expect messages not displayed

baltazarortiz, 2019-11-22:
From the example output in [Using The PeekPokeTester](https://github.com/freechipsproject/chisel-testers/wiki/Using%20the%20PeekPokeTester), it appears that there should be messages for PEEK/POKE/STEP/EXPECT calls. However, even when I run the GCD test being used in the writeup, I don't see any of these messages.

Is this intended behavior now, or is it possible to enable these messages somehow?

My build.sbt contains `  val defaultVersions = Map(                                                      
    "chisel3" -> "3.1.+",                                                         
    "chisel-iotesters" -> "1.2.+"                                                 
    )   `. If there is more relevant info needed, let me know!

seldridge, 2019-11-22:
For your default versions, can you bump to the newest versions and verify if this is still an issue?

```scala
val defaultVersions = Map( "chisel3" -> "3.2.+", "chisel-iotesters" -> "1.3.+" )
```

baltazarortiz, 2019-11-22:
Same issue after updating - I ran ` ./run-examples.sh GCD` and get

```
[info] Done compiling.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.protobuf.UnsafeUtil (file:/home/baltazar/.sbt/boot/scala-2.12.4/org.scala-sbt/sbt/1.1.1/protobuf-java-3.3.1.jar) to field java.nio.Buffer.address
WARNING: Please consider reporting this to the maintainers of com.google.protobuf.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[info] Packaging /home/baltazar/chisel-tutorial/target/scala-2.11/chisel-tutorial_2.11-3.1.0.jar ...
[info] Done packaging.
[info] Compiling 63 Scala sources to /home/baltazar/chisel-tutorial/target/scala-2.11/test-classes ...
[info] Done compiling.
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Packaging /home/baltazar/chisel-tutorial/target/scala-2.11/chisel-tutorial_2.11-3.1.0-tests.jar ...
[info] Done packaging.
[info] Running examples.Launcher GCD
Starting tutorial GCD
[info] [0.003] Elaborating design...
[info] [1.147] Done elaborating.
Total FIRRTL Compile Time: 636.6 ms
file loaded in 0.102913863 seconds, 24 symbols, 19 statements
[info] [0.002] SEED 1574395807478
test GCD Success: 3 tests passed in 29 cycles in 0.029951 seconds 968.24 Hz
[info] [0.013] RAN 24 CYCLES PASSED
Tutorials passing: 1
[success] Total time: 47 s, completed Nov 22, 2019, 4:10:10 AM
```

seldridge, 2019-11-22:
Got it! Thanks for updating with your output here.

There's an option al to turn on verbose printing.

It looks like you're using the `chisel-tutorial`, so I would change [this line](https://github.com/ucb-bar/chisel-tutorial/blob/release/src/test/scala/examples/GCDTests.scala#L38) to be:

```diff
-Driver(() => new GCD, backend)((c) => new GCDTests(c)) should be (true)
+Driver(() => new GCD, backend, true)((c) => new GCDTests(c)) should be (true)
```

[The `apply` method of the `chisel3.iotesters.Driver` object](https://www.chisel-lang.org/api/chisel-testers/latest/chisel3/iotesters/Driver$.html#apply[T%3C:chisel3.MultiIOModule](dutGen:()=%3ET,backendType:String,verbose:Boolean,testerSeed:Long)(testerGen:T=%3Echisel3.iotesters.PeekPokeTester[T]):Boolean) takes a third argument, `verbose`, that *should* turn on the behavior you see in the iotesters wiki.

Let me know if that does it for you.

Unrelated comment, the Chisel tutorial is quite old at this point. If you're looking to ramp up on Chisel, you *may* have a better experience using the [Chisel bootcamp](https://github.com/freechipsproject/chisel-bootcamp) which you can either run locally or [in your browser](https://mybinder.org/v2/gh/freechipsproject/chisel-bootcamp/master) (with the latter option not allowing you to save your work and it kicking you out if you idle for something like 30 minutes). Disregard if you're aware of all this.

baltazarortiz, 2019-11-22:
No problem! The extra argument unfortunately didn't work for me (same output as before) - I believe I tried it in my personal project and it wasn't working, which is why I decided to test it on the tutorial to see if there was something I had configured incorrectly.

I have gone through the Chisel bootcamp, I've just been using tutorial for testing SBT things and for examples of more testbenches and things :)

seldridge, 2019-11-22:
Cool. Yeah, this warrants a bit more investigation... Sorry about the issues.

One thing I did get to work is playing with the example in the Chisel template (which is supposed to be a jumping off point for post-bootcamp work and/or a basic starting point for a Chisel project). Running that locally, I was able to turn the verbose printing on and off with the presence/absence of the `--is-verbose` argument. (Oddly, the third argument of `Driver.apply` did exhibit the correct behavior for me there.)

This is the specific test that I'm talking about: https://github.com/freechipsproject/chisel-template/blob/release/src/test/scala/gcd/GCDUnitTest.scala#L94

You may try also playing with that and see if you can triage the other.

baltazarortiz, 2019-11-22:
Should the verbose messages be appearing on the command line, or hidden in a log file somewhere?

I ran `sbt 'testOnly gcd.GCDTester -- -z Basic'` in a newly cloned copy of the chisel template and my output matches the results in the [Did It Work](https://github.com/freechipsproject/chisel-template#did-it-work) section. I get the `running with --is-verbose` message but no additional poke/expect messages like I would expect from looking at the unit test code.

baltazarortiz, 2019-12-09:
Forgot to update earlier, but once I changed added `--is-verbose` to my `Driver.execte()` call the verbose messages are showing up now. Not sure why they weren't before, but I'll close this since it seems to be working correctly.

The full line I am using that works: 

`chisel3.iotesters.Driver.execute(Array("--is-verbose", "--backend-name", "verilator","--fint-write-vcd")`

closed 2019-12-09 by baltazarortiz