freechipsproject/chisel-testers #272: firrtl vs. verilator behave different

ShuyunJia, 2019-11-27:
My test code is :

```
import chisel3._
import chisel3.util._
import chisel3.iotesters
import chisel3.iotesters.{ChiselFlatSpec, Driver, PeekPokeTester}

class MyDut extends Module {
  val io = IO(Flipped(Decoupled(UInt(8.W))))
  val counter = RegInit(0.U(8.W))
  when(io.valid){  counter := counter +1.U }
  when(io.fire()){
    printf("=======io fire=======\n")
  }
  io.ready := counter >= 4.U
}
class MyDutUnitTester(c: MyDut) extends PeekPokeTester(c) {
  val myDut = c
  poke(myDut.io.valid, 1)
  poke(myDut.io.bits, 4)
  while(peek(myDut.io.ready) == 0){
    step(1)
  }
  step(1)
  poke(myDut.io.valid, 0)
  step(1)
}

class MyDutTester extends ChiselFlatSpec {
  it should "print only once fire" in {
    iotesters.Driver.execute(Array("--backend-name", "verilator"), () => new MyDut) {
      c => new MyDutUnitTester(c)
    } should be (true)
  }

  it should "print twice fire" in {
    iotesters.Driver.execute(Array("--backend-name", "firrtl"), () => new MyDut) {
      c => new MyDutUnitTester(c)
    } should be (true)
  }
}
```

With the `firrtl` backend, I got `=====io fire=====` logged twice while the `verilator` backend only generate once. I think `verilator` is correct here. 

It seems to be the similiar with #124. 

The waveform is: 
[image]

At the marked clock edge, the ready signal is derived from flip-flop `counter`, thus would not satisfy setup-hold time. However, the `firrtl` backend execute the print this cycle. 

The difference is when to calculate `io.fire()`. It should calculate `io.fire()` at the same time with other flip-flopps, which means it should also satisfy the setup-hold time requirement.

seldridge, 2019-11-27:
Does the same behavior happen with the Treadle backend? For more information, Treadle is FIRRTL interpreter's replacement and should generally always be used instead of the interpreter.

ShuyunJia, 2019-11-27:
Treadle prints twice `io fire` in my case.

chick, 2019-12-03:
@chick ToDo see what VCS does

albert-magyar, 2019-12-04:
Resolution after talking to @chick : the behavior of `printf` enables will be modified in Treadle to resolve this issue.

chick, 2019-12-06:
@ShuyunJia I have created [Treadle #159](https://github.com/freechipsproject/treadle/pull/159) that I hope will fix your problem, though there is still one problematic edge case involving reset.
If you have a chance to test with this branch let me know how it goes, thanks

ShuyunJia, 2019-12-09:
@chick Sorry for the delay. I tested https://github.com/freechipsproject/treadle/pull/159# with my case, now treadle generates the same result as verilator.

closed 2021-03-04 by ekiwi