freechipsproject/chisel-testers #311: Removed `io` in Module cause compilation error

SihaoLiu, 2021-01-30:
I checked the 2.5-release branch and try to publish it to Local, but it gave me compilation error says:

```
...
[error] .chisel-testers/src/main/scala/chisel3/iotesters/OrderedDecoupledHWIOTester.scala:394:48: value io is not a member of chisel3.Module
[error]     io_info = new IOAccessor(device_under_test.io)
[error]                                                ^
[error] ./chisel-testers/src/main/scala/chisel3/iotesters/SteppedHWIOTester.scala:173:48: value io is not a member of chisel3.Module
[error]     io_info = new IOAccessor(device_under_test.io)
[error]                                                ^
...
```
I realized that it is caused by removed io in Module ([#1745](https://github.com/chipsalliance/chisel3/pull/1745)), I would like to know how to fix it, thank you.

sequencer, 2021-01-30:
Fixed in #309

SihaoLiu, 2021-01-30:
I see, thanks

closed 2021-02-01 by SihaoLiu